tinygo-org/tinygo #1544: GBA: interrupt handlers are not called

mogenson, 2020-12-23:
I've been unable to get interrupt handlers working on the gameboy-advance target. Below is a minimal example that changes the color of a pixel based on A or B button presses.

```
package main

import (
	"image/color"
	"machine"
	"runtime/interrupt"
	"runtime/volatile"
	"unsafe"
)

var KEYPAD = (*volatile.Register16)(unsafe.Pointer(uintptr(0x04000130)))
var KEYCNT = (*volatile.Register16)(unsafe.Pointer(uintptr(0x04000132)))
var display = machine.Display

func main() {
	display.Configure()
	display.SetPixel(120, 80, color.RGBA{R: 255})
	KEYCNT.Set(0x0003) // generate interrupts for A and B buttons
	interrupt.New(machine.IRQ_KEYPAD, keypadIrq).Enable()

	for {
		//readKeypad()
	}
}

func readKeypad() {
	if KEYPAD.Get()&1 == 0 {
		// A is pushed when bit 0 is cleared
		display.SetPixel(120, 80, color.RGBA{G: 255})
	} else if KEYPAD.Get()&2 == 0 {
		// B is pushed when bit 1 is cleared
		display.SetPixel(120, 80, color.RGBA{B: 255})
	}
}

func keypadIrq(interrupt.Interrupt) {
	readKeypad()
}
```

This builds and runs fine when polling for button pushes with `readKeypad()`. The `KEYCTL`, `IE`, `IF`, and `IME` registers all seem to be set correctly, according to the `mgba` emulator.

Using GDB I can get a breakpoint on [handleInterrupt()](https://github.com/tinygo-org/tinygo/blob/15361c829e2e1cb559a04e2b1a66bef0416a48ee/src/runtime/interrupt/interrupt_gameboyadvance.go#L23), but `callInterruptHandler()` doesn't seem to go anywhere.

Could this be an issue with switching from system mode to user mode, or with the change from Thumb2 to ARM?

deadprogram, 2020-12-26:
Does your branch have the recent commits from https://github.com/tinygo-org/tinygo/pull/1430 in it? If so, can you try without it?

mogenson, 2020-12-26:
I've tried with and without #1430 (ie. dev and release) with the same result.

I'm unfamiliar with the structure of a TinyGO binary, but I can't find a symbol for my `keypadIrq` interrupt handler in the generated elf file.

```
$ nm keypad.elf
08000110 t $a.0
08000274 t $a.0
08000000 t $a.0
08000190 t $a.10
08000198 t $a.14
080001e8 t $a.16
08000228 t $a.18
080000c0 t $a.2
0800022c t $a.23
0800018c t $d.1
08000004 t $d.1
080001d4 t $d.15
08000218 t $d.17
08000270 t $d.24
080000fc t $d.3
0800022c T command-line-arguments.main
03000c14 B _ebss
03000c00 D _edata
03000c14 B _globals_end
03000c00 D _globals_start
08000110 T handleInterrupt
080000f4 t handleInterruptARM
02040000 A _heap_end
02000000 A _heap_start
08000190 T main
08000274 T memset
03000c10 b runtime.endBlock
03000c0c b runtime.nextAlloc
03000c11 b runtime.poolStart
08000198 t runtime.preinit
080001e8 t runtime.run
08000228 t runtime.run$1
03000c00 b runtime.runqueue
03000c12 b runtime.stackOverflow
03000c08 b runtime.zeroSizedAlloc
03000c00 B _sbss
03000c00 D _sdata
0800028c A _sidata
00000400 A __stack_size_irq
00000800 A __stack_size_usr
03000400 T _stack_top
03000000 T _stack_top_irq
08000000 T _start
080000c0 t start_vector
```

aykevl, 2020-12-26:
Have you tried any other interrupt, for example the V-blank interrupt? That could help narrow down the issue.

Also, you might want to put something in here, at the end of `main`:

```go
	for {
		//readKeypad()
	}
```

Anything that's not a no-op, really (such as drawing to the display even if it's always the same location). The reason is that LLVM currently might consider endless loops to be undefined behavior, which it is in some cases in C. The Rust project has a similar issue where infinite loops are sometimes incorrectly optimized away. Putting something in there should make sure it doesn't happen here.

> I'm unfamiliar with the structure of a TinyGO binary, but I can't find a symbol for my `keypadIrq` interrupt handler in the generated elf file.

This is normal. It is likely the `keypadIrq` function is inlined in the interrupt handler.

mogenson, 2020-12-27:
I guess there is something wrong with my keypad example. I've been unable to get any `IRQ_KEYPAD` or `IRQ_COM` interrupts to fire (even when manually setting the IF bits in the emulator).

However, I tried `IRQ_VBLANK` and `IRQ_TIMER0`. These seem to work fine. There are some examples below.

```
package main

import (
	"image/color"
	"machine"
	"runtime/interrupt"
	"runtime/volatile"
	"unsafe"
)

var display = machine.Display
var DISPSTAT = (*volatile.Register16)(unsafe.Pointer(uintptr(0x4000004)))

func main() {
	display.Configure()
	display.SetPixel(80, 80, color.RGBA{R: 255})
	DISPSTAT.SetBits(1 << 3) // interrupt on vblank
	interrupt.New(machine.IRQ_VBLANK, vblankIrq).Enable()

	for {
		display.SetPixel(100, 80, color.RGBA{G: 255})
	}
}

func vblankIrq(interrupt.Interrupt) {
	display.SetPixel(120, 80, color.RGBA{B: 255})
}
```

```
package main

import (
	"image/color"
	"machine"
	"runtime/interrupt"
	"runtime/volatile"
	"unsafe"
)

var display = machine.Display
var TM0CNT_H = (*volatile.Register16)(unsafe.Pointer(uintptr(0x4000102)))

func main() {
	display.Configure()
	display.SetPixel(80, 80, color.RGBA{R: 255})
	TM0CNT_H.SetBits(1 << 6 | 1 << 7) // start timer, interrupt on overflow
	interrupt.New(machine.IRQ_TIMER0, timerIrq).Enable()

	for {
		display.SetPixel(100, 80, color.RGBA{G: 255})
	}
}

func timerIrq(interrupt.Interrupt) {
	display.SetPixel(120, 80, color.RGBA{B: 255})
}
```

You're right. My IRQ handler is inlined and the symbol is not present. Adding or removing dummy work in the `for {}` loop doesn't seem to make a difference in any of the examples.

aykevl, 2020-12-27:
Interesting. Maybe you need to set or clear some other bit? Interrupts usually need to have multiple things configured correctly for them to work.

mogenson, 2020-12-27:
OMG, I was going off of [this documentation](https://www.cs.rit.edu/~tjh8300/CowBite/CowBiteSpec.htm#Keypad%20input%20and%20control%20registe) and completely missed bit 14 of `KEYCNT`.

Interrupts work fine. I need to look at the serial communication registers. I now assume I must have misconfigured them as well.

Closing this issue, sorry for the noise!

closed 2020-12-27 by mogenson

aykevl, 2020-12-27:
Glad you found the solution!