tinygo-org/tinygo #1551: Move to Rust SVD library for STM32

kenbell, 2020-12-29:
The discussion here https://github.com/tinygo-org/cmsis-svd/pull/7 concluded we could look at moving TinyGo to use the rust SVD library (https://github.com/stm32-rs/stm32-rs) for STM32 devices.

A first pass implementation is implemented in this PR: https://github.com/tinygo-org/tinygo/pull/1550

aykevl, 2020-12-29:
@kenbell as I indicated, I was already working on this feature. I've made somewhat similar changes to gen-device-svd, but merging all enum fields (with the same value) to a single field instead of duplicating them with added suffixes. I'm also still working on fixing stm32 support to be able to use this new repository: some changes are needed in the runtime and machine packages.

> This change requires some environment changes to generate the device/stm32 source:
> 
>    * python3 to be installed
>    * svdtools to be installed (`pip3 install --upgrade --user svdtools`)

I'm working on creating a repository with just the patched SVD files, avoiding these build dependencies (which I would really like to avoid, they will complicate things).

kenbell, 2020-12-29:
That's cool - this change is letting me keep trucking and raise PRs against stm32-rs for now.  Avoiding python3 and svdtools dependency does sound like a better approach.

aykevl, 2021-01-05:
I have made a PR here: https://github.com/tinygo-org/tinygo/pull/1557

It does a lot more, it also removes the handcrafted bitifield files from src/device/stm32 and in that process fixes some bugs (hopefully). But I've tried to avoid any other changes.

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram