tinygo-org/tinygo #1552: Build fails for Teensy 3.6 when fmt is used

firelizzard18, 2020-12-29:
* OS: macOS Catalina
* LLVM: `make llvm-project && make llvm-build` (tinygo-org/llvm-project@9ecb19f774994a3efff5a6b89aa43ba2b8d2dd23)
* TinyGo: `make` (fb0bb69f622c94e6aa2e6e340faa0ba2146cf440)

```golang
package main

import "fmt"

func main() {
	fmt.Sprintf("Hello %s!", "world")
}
```

```shell
$ ../tinygo/build/tinygo build -target feather-m4 -o main.hex main.go
# success
$ ../tinygo/build/tinygo build -target teensy40 -o main.hex main.go
# success
$ ../tinygo/build/tinygo build -target teensy36 -o main.hex main.go
error: could not parse executable for stack size analysis: could not decode .debug_frame bytecode op 0x5 (for address 0x97e)

aykevl, 2021-01-05:
Interesting bug, I can reproduce it.

aykevl, 2021-01-06:
https://github.com/tinygo-org/tinygo/pull/1564 should fix this.

firelizzard18, 2021-01-10:
👍 #1564 fixes this for me

closed 2021-01-10 by firelizzard18