tinygo-org/tinygo #1553: Very slow compared others?

kokizzu, 2020-12-29:
Tried to compile and run this code, but it doesn't complete within 1000s (while other languages/compiler completed within 1-10s)

```
// map.go
package main

import (
        "bytes"
        "fmt"
        "strconv"
)

type MSD map[string]float64

const MAX_DATA = 1230000

var i2ch []byte

func init() {
        i2ch = []byte{'0', '1', '2', '3', '4', '5', '6', '7', '8', '9', 'a', 'B', 'c', 'D', 'e', 'F'}
}
func get_first_digit(d float64) int {
        for d > 10 {
                d /= 10
        }
        return int(d)
}
func to_rhex(v int) string {
        hex := bytes.Buffer{}
        for v > 0 {
                hex.WriteByte(i2ch[v%16])
                v /= 16
        }
        return hex.String()
}
func set_or_inc(m map[string]float64, key string, set, inc int, ctr *int) {
        if _, ok := m[key]; !ok {
                m[key] = float64(set)
        } else {
                m[key] += float64(inc)
                *ctr += 1
        }
}
func main() {
        m := MSD{}
        dup1, dup2, dup3 := 0, 0, 0
        for z := MAX_DATA; z > 0; z-- {
                val2 := MAX_DATA - z
                val3 := MAX_DATA*2 - z
                key1 := strconv.Itoa(z)
                key2 := strconv.Itoa(val2)
                key3 := to_rhex(val3)
                set_or_inc(m, key1, z, val2, &dup1)
                set_or_inc(m, key2, val2, val3, &dup2)
                set_or_inc(m, key3, val3, z, &dup3)
        }
        fmt.Println(dup1, dup2, dup3)
        total, verify, count := 0, 0, 0
        for k, v := range m {
                total += get_first_digit(v)
                verify += len(k)
                count += 1
        }
        fmt.Println(total, verify, count)
}
```
[image]

kokizzu, 2020-12-29:
I'm using ubuntu 20.04 64-bit, default target

kokizzu, 2020-12-29:
sometimes segfault:

```
$ time tinygo run map.go             
error: failed to run compiled binary /tmp/tinygo923402887/main: signal: segmentation fault (core dumped)
Command exited with non-zero status 1

CPU: 159.84s    Real: 161.43s   RAM: 320144KB
```

aykevl, 2020-12-29:
Maps are not at all optimized in TinyGo, they have O(n²) behavior. This should be improved, the current implementation was only written to be compatible with existing programs that use maps.

> sometimes segfault:

Interesting. Maybe @niaow can comment on this?

niaow, 2021-01-01:
Hmm. . . I don't seem to be able to replicate this so far. I will let this run for a few more hours, but it seems like it should have crashed by now.

FrankReh, 2021-05-04:
For anyone else running into this thread, it is possible to make the tinygo map very efficient - and the tinygo wasm code runs faster than the go wasm code. Make the map with a good size hint and consider allocating enough heap to begin with.

The current implementation won't increase the bucket width, just the depth of bucket chains as more entries overflow the preallocated buckets. If one doesn't provide a size width, the map is built with a single bucket and then one gets the worst performance, the hash of the key is effectively ignored and all entries are added to an ever increasing linked list.

Each bucket holds 8 entries. 8K entries for example would all hash to the same bucket and then a chain of 1K buckets would have to be created. If testing with 1.2 million entries, the chain will have 150,000 buckets. But the real point is to add
another entry to a map with 1.2 million entries, all 1.2 million entry keys have to be tested for equality first before the map can decide that it is not a duplicate. So create the map large enough to begin with. Or in an application, watch the size and if it grows larger than expected, consider building a new larger map and copying from the smaller one. That's effectively what the go map does under the covers.

With a properly sized map, and with enough headroom in the heap, the tinygo wasm code runs faster than the go wasm code.
1.3 seconds vs 1.9 seconds on one amd64 machine I test on. The code I used is below.

Also know that the go wasm binary starts with a 4GB wasm memory block while the tinygo wasm binary starts with a 128KB wasm memory block.

Without the tweak of the map size hint, and getting the wasm memory to be large enough up front, the go wasm code runs in 3.5s and the tinygo wasm won't finish because of the O(N*N) nature of the single bucket map and the time spent in running GC cycles.

Curious how the same code runs compiled with go for the native OS? 0.7s. Almost twice as fast as the tinygo wasm code running in a nodejs vm.

FrankReh, 2021-05-04:
``` go
package main

import (
	"runtime"
	"strconv"
	"strings"
	"time"
)

const MAX_DATA = 1_230_000
const BallastCount = 20_000_000
const MapSizeHint = 3 * MAX_DATA

const i2ch = "0123456789aBcDeF"

func get_first_digit(d float64) int {
	for d > 10 {
		d /= 10
	}
	return int(d)
}
func to_rhex(v int) string {
	hex := new(strings.Builder)
	for v > 0 {
		hex.WriteByte(i2ch[v%16])
		v /= 16
	}
	return hex.String()
}
func set_or_inc(m map[string]float64, key string, set, inc int, ctr *int) {
	if _, ok := m[key]; !ok {
		m[key] = float64(set)
	} else {
		m[key] += float64(inc)
		*ctr += 1
	}
}
func main() {
	ballast(BallastCount)
	runtime.GC()
	println("ballast done")
	start := time.Now()
	m := make(map[string]float64, 3*MAX_DATA)
	dup1, dup2, dup3 := 0, 0, 0
	for z := MAX_DATA; z > 0; z-- {
		val2 := MAX_DATA - z
		val3 := MAX_DATA*2 - z
		key1 := strconv.Itoa(z)
		key2 := strconv.Itoa(val2)
		key3 := to_rhex(val3)
		set_or_inc(m, key1, z, val2, &dup1)
		set_or_inc(m, key2, val2, val3, &dup2)
		set_or_inc(m, key3, val3, z, &dup3)
	}
	dur := time.Since(start)
	println(dup1, dup2, dup3)
	total, verify, count := 0, 0, 0
	for k, v := range m {
		total += get_first_digit(v)
		verify += len(k)
		count += 1
	}
	println(total, verify, count)
	println("MAX_DATA:", MAX_DATA, "BallastCount:", BallastCount, "MapSizeHint:", MapSizeHint, "Loop time:", dur.String())
}

type Link struct {
	*Link
}

func ballast(count int) {
	var link *Link
	for i := 0; i < count; i++ {
		link = &Link{link}
	}

	// Now walk the chain and zero each link. This handles both the case of
	// this code being inlined with the step running the GC and the case where
	// some mark false positives would keep some block from being swept back to
	// the heap.

	for link != nil {
		t := link
		link = link.Link
		t.Link = nil
	}
}
```

aykevl, 2021-05-05:
> sometimes segfault:
> 
> ```
> $ time tinygo run map.go             
> error: failed to run compiled binary /tmp/tinygo923402887/main: signal: segmentation fault (core dumped)
> Command exited with non-zero status 1
> 
> CPU: 159.84s    Real: 161.43s   RAM: 320144KB
> ```

I suspect this crash is related to this issue in the extalloc GC: https://github.com/tinygo-org/tinygo/pull/1445#issuecomment-832300285. I've seen memory related crashes before with the extalloc that were worked around with `-gc=conservative`.
You can try compiling with `-gc=conservative` to see whether that fixes it.

dgryski, 2022-02-25:
`extalloc` has been removed; I don't see any more memory corruption crashes.

I agree the maps could use growing logic, so we should probably retitle this bug.

dgryski, 2022-03-08:
With https://github.com/tinygo-org/tinygo/pull/2687 this now takes ~5s (or sometimes ~18-20s if the garbage collector isn't feeling nice.)

deadprogram, 2022-04-28:
This should now have been addressed thanks to @dgryski work on https://github.com/tinygo-org/tinygo/pull/2755

If so we can mark this issue for closing on next release.

commit 7e05efa2 (2022-04-04): src/runtime: first darft of map growth code

Fixes #1553
PR #2755 (2022-04-28): src/runtime: second draft of map growth code
Fixes #1553
commit 6812a4dc (2022-04-28): src/runtime: first darft of map growth code

Fixes #1553
PR #2740 (2022-06-17): src/runtime: switch to stronger hash function
Fixes #1553
closed 2022-04-28 by deadprogram