tinygo-org/tinygo #1555: 怎么样提交一个新板子   芯片用的是stm32f103c8t6

zhuqiyun, 2021-01-01:
[image]
[image]

deadprogram, 2021-01-02:
您好，谢谢您的提问。

不好意思的翻译表示歉意。 我不得不使用谷歌。

有关如何为TinyGo贡献新的董事会定义的信息，请参见此处的Wiki部分：
https://github.com/tinygo-org/tinygo/wiki/Adding-a-new-board#adding-a-new-board

您可以在此处找到有关如何捐款的一般信息：
https://github.com/tinygo-org/tinygo/blob/release/CONTRIBUTING.md#some-specific-hardware-you-want-to-use-does-not-appear-to-be-in-tinygo

谢谢！

davecheney, 2026-10-24:
Thank you for your interest in contributing board support to TinyGo. The reply above provides the board and contribution instructions, so I’m closing this answered question. Please submit a pull request when your board definition is ready.

closed 2026-10-24 by davecheney