tinygo-org/tinygo #1559: Import uses the wrong module when function name is "write".

ckaznocha, 2021-01-05:
When setting the target as WASI specifying a module to import a function named `write`, the generated wasm code will import `"wasi_snapshot_preview1" "fd_write"` rather than a function called `write` from the correct module.

Example go program with the incorrect import:

```Go
package main

//go:wasm-module my_module
//export foo
func foo() int // <-- Works correctly

//go:wasm-module my_module
//export write
func bar() int // <-- imports fd_write from wasi_snapshot_preview1

func main() {
    foo()
    bar()
}
```

Compiled using `tinygo build -target=wasi -scheduler=none -wasm-abi=generic -o ./module.wasm main.go`

Result after converting main.wasm to wat:

```wat
(module
  (type (;0;) (func (result i32)))
  (type (;1;) (func (param i32 i32 i32 i32) (result i32)))
  (type (;2;) (func (param i32 i32 i32) (result i32)))
  (type (;3;) (func))
  (type (;4;) (func (param i32 i32)))
  (import "my_module" "foo" (func $foo (type 0)))
  (import "wasi_snapshot_preview1" "fd_write" (func $__wasi_fd_write (type 1)))
  (func $memset (type 2) (param i32 i32 i32) (result i32)
    (local i32 i32 i32 i64)
    block  ;; label = @1
      local.get 2
      i32.eqz
      br_if 0 (;@1;)
      local.get 0
      local.get 1
      i32.store8
      local.get 2
      local.get 0
      i32.add
      local.tee 3
      i32.const -1
      i32.add
      local.get 1
      i32.store8
      local.get 2
      i32.const 3
      i32.lt_u
      br_if 0 (;@1;)
      local.get 0
      local.get 1
      i32.store8 offset=2
      local.get 0
      local.get 1
      i32.store8 offset=1
      local.get 3
      i32.const -3
      i32.add
      local.get 1
      i32.store8
      local.get 3
      i32.const -2
      i32.add
      local.get 1
[... 716 lines cut ...]
    i32.shl
    i32.const 65536
    i32.sub
    i32.const 6
    i32.shr_u
    call $memset
    drop
    local.get 0
    local.get 0
    call $command-line-arguments.main)
  (func $command-line-arguments.main (type 4) (param i32 i32)
    (local i32)
    call $foo
    drop
    local.get 2
    local.get 2
    call $command-line-arguments.write
    drop)
  (table (;0;) 1 1 funcref)
  (memory (;0;) 1)
  (global (;0;) (mut i32) (i32.const 65536))
  (export "memory" (memory 0))
  (export "_start" (func $_start))
  (export "command-line-arguments.main" (func $command-line-arguments.main)))
```

aykevl, 2021-01-05:
Well that's interesting. I suspect there is some interaction with wasi-libc. This might mean you cannot give a function a name that matches a libc name such as `write` because those names will link against wasi-libc instead of leaving them undefined during linking.

_Maybe_ this is a wasi-ld bug, I'm not sure what the intended behavior is when the module name differs.

fgsch, 2021-02-21:
Function names are per module so it should be possible to have multiple `write` across different imports.

#1643 might be related.

deadprogram, 2021-03-03:
Now that #1661 has been merged, can someone please re-test this issue against the `dev` branch? Thank you.

fgsch, 2021-03-04:
Confirmed. Revision dcff8b6 produces the expected imports.

fgsch, 2021-03-15:
@deadprogram not sure if you are waiting for the reporter or someone else to test this but in my tests this is fixed.

deadprogram, 2021-03-18:
Thanks @fgsch just tagged with label for next release, at which point it will get closed.

fgsch, 2021-03-18:
@deadprogram ok. fwiw, the fix is already included in 0.17.0.

deadprogram, 2021-03-18:
AH, in that case, closing. Thanks for clarifying that @fgsch

PR #1661 (2021-03-03): Namespaced Wasm Imports so they don't conflict across modules, or reserved LLVM IR
closes #1643
closes #1559

Kind of relates to #1647
Doesn't fix, but puts on the right path I think, for relates to #1648

So this PR was opened from a discussion in the tinygo slack: https://gophers.slack.com/archives/CDJD3SUP6/p1614201204223100 . Involving a conflict with wasm imports with the name of `write`. Which conflicts with the LLVM IR `@write`, and requires decoupling the LLVM IR symbol, and the `wasm-import-name`.

I tested this pr with the following code: https://github.com/torch2424/tinygo-compiler- [... cut]
closed 2021-03-18 by deadprogram