tinygo-org/tinygo #1566: how use embedded eeprom on the arduino uno with tinygo?

AndreiLosev, 2021-01-06:
how use embedded eeprom on the arduino uno with tinygo?

aykevl, 2021-01-06:
Unfortunately we do not yet support EEPROM, hopefully we'll add support for it in the future.

SoleimanyBen, 2022-11-12:
@aykevl any updates on this feature? I'm willing to take a look at implementing this if I can get some insight on what would needed to be done. From the limited research (and understanding I have of the internals) I've done, implementing the EEPROM lib would require reading and writing to specific EEPROM registers that are on the CPU? It also appears that these registers are already implemented so it would just require modifying the CPU specific `device` package file?

For example, with something like the ATMega8/16/32u2, it would require implementing a struct that interacts with `EEARL`, `EEARH`, `EEDR`, and `EECR`. If this general logic is correct, I don't mind attemtping to implement a general EEPROM library for interacting with the CPUs embedded memory.

https://ww1.microchip.com/downloads/en/DeviceDoc/doc7799.pdf look at section 7.5

aykevl, 2022-11-17:
Your help in getting this implemented would be appreciated!
As a first step, do you have a proposed API for EEPROMs? While they appear to be less common on newer Cortex-M chips, I'm sure there are still enough of them that we need to make a generic API for this.

> It also appears that these registers are already implemented so it would just require modifying the CPU specific `device` package file?

To clarify, the registers live in device/avr. An implementation would be for the machine package.

deadprogram, 2022-11-17:
Might want to use same API as https://github.com/tinygo-org/drivers/blob/release/at24cx/at24cx.go

SoleimanyBen, 2022-11-18:
> As a first step, do you have a proposed API for EEPROMs? 

I was thinking of using a similar API to what @deadprogram posted.

```golang
type EEPROM struct {
}

func (e *EEPROM) ReadByte(addr uint16) (uint8, error)
func (e *EEPROM) ReadAt(addr uint16, buf []byte) (int, error)
func (e *EEPROM) WriteByte(addr uint16, value byte) error
func (e *EEPROM) WriteAt(addr uint16, data []byte) (int, error)
func (e *EEPROM) Seek(offset int64, whence int) (int64, error)
func (e *EEPROM) Write(data []byte) (int, error)
func (e *EEPROM) Read(data []uint8) (n int, err error)
```

aykevl, 2022-11-19:
Hmm, not entirely sure about that API to be honest.
- The last 3 methods require a seek offset, which is state that must be kept by the implementation. I think it's best to avoid this state.
- `ReadAt` and `WriteAt` don't implement [`ReaderAt`](https://pkg.go.dev/io#ReaderAt) and [`WriterAt`](https://pkg.go.dev/io#WriterAt) - the addr and buf parameters are reversed and the addr parameter is `uint16` instead of `int64`
- `ReadByte` and `WriteByte` don't implement io.ByteReader and io.ByteWriter.

But of course, EEPROM is typically small and so a 64-bit address is kind of overkill. 16-bit may be enough for most (although there do exist EEPROM chips that are bigger than 64kB). But breaking compatibility with these interfaces doesn't seem like a good idea to be honest.

Here is a suggestion:

```go
type EEPROM struct {
}

func (e *EEPROM) ReadByteAt(addr int64) (byte, error)
func (e *EEPROM) ReadAt(buf []byte, off int64) (int, error)
func (e *EEPROM) WriteByteAt(value byte, addr int64) error
func (e *EEPROM) WriteAt(data []byte, off int64) (int, error)
```

`ReadAt` and `WriteAt` implement io.ReaderAt and io.WriterAt. `ReadByteAt` and `WriteByteAt` are like `ReadByte` and `WriteByte` but use an address offset.

Let me know what you think :)

SoleimanyBen, 2022-11-21:
Great, thank you for the insight! :)

I like the API changes, so I'll go ahead and refactor what I already have to reflect this.

SoleimanyBen, 2022-11-21:
would you happen to know why the bitfields for the EEPROM are not accurate? They all seem to be +1 more than they should be?

For example, all of these should be -1
```golang
// Bitfields for EEPROM: EEPROM
const (
	...
	EECR_EEPE      = 0x2  // EEPROM Write Enable
	EECR_EEPE_Msk  = 0x2  // EEPROM Write Enable
	EECR_EERE      = 0x1  // EEPROM Read Enable
	EECR_EERE_Msk  = 0x1  // EEPROM Read Enable
)
```

aykevl, 2022-12-08:
Those bits look correct. Why would they be -1? Note that `EECR_EERE` for example is a bit mask, so that's why it's not 0. 0x1 is 0b0000_0001, 2 is 0b0000_0010, etc.