tinygo-org/tinygo #1568: feather-stm32f405: I2C not working

deadprogram, 2021-01-10:
I finally acquired an Adafruit feather-stm32f405 board. Flashed it with the latest TInyGo `dev` branch. The MPU6050 example from here does not run correctly: https://github.com/tinygo-org/drivers/blob/release/examples/mpu6050/main.go

It only returns zero values...

@ardnew any thoughts?

aykevl, 2021-01-10:
Did you try the dev branch before #1557? It may have broken things.

deadprogram, 2021-01-10:
Same behavior in current `release` branch.

deadprogram, 2021-02-27:
I wonder if the change similar to https://github.com/tinygo-org/tinygo/commit/66f76833ad0ac1dd31befa3ecbd8c0d1e5bb5ca3 would address the issue I have with the stm32f405? Seems like a good chance it would.

deadprogram, 2021-02-27:
OK I can confirm that now with https://github.com/tinygo-org/tinygo/commit/66f76833ad0ac1dd31befa3ecbd8c0d1e5bb5ca3 in the `dev` branch that if I add a `time.Sleep(100 * time.Millisecond)`  to the MPU6050 example as suggested by @kenbell this works:

```go
func main() {
	machine.I2C0.Configure(machine.I2CConfig{})

	time.Sleep(100 * time.Millisecond)
	accel := mpu6050.New(machine.I2C0)
	accel.Configure()
...
```

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

PR #1609 (2021-02-16): STM32 I2C: replace addressable interface with simpler type
The `addressable` interface type is unnecessary and can be removed, which simplifies some of the code. This doesn't resolve #1568, but it's a start with diagnosing the issue.
closed 2021-03-08 by deadprogram