tinygo-org/tinygo #1570: Add -verbose flag to enable external commands logging (eg: openocd)

ofauchon, 2021-01-13:
Hi all, 

I could not find a (clean) way  to display the external commands and arguments executed by tinygo (eg: openocd  )
Is there one ? 

At the moment, I'm using "strace -f tinygo .... "  to get the information (which is an ugly method)

So I made a patch to add a '-verbose' CLI option that displays the external executed commands. 

See here for the implementation: 

https://github.com/tinygo-org/tinygo/compare/dev...ofauchon:verbose_execs

Please tell me if you like (or not) this small contribution so I create a Pull Request. 

Olivier

aykevl, 2021-01-13:
I think this is a good idea. However, I think the flag should be named `-x` to match the Go toolchain.

ofauchon, 2021-01-13:
Thanks, 

Using "-x" makes more sense ... 

Pull Request is here: https://github.com/tinygo-org/tinygo/pull/1572

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram