tinygo-org/tinygo #1573: Source install error : "ld: library not found for -llibxml2.tbd"

DDkuang, 2021-01-15:
system info : macOS Big Sur
CPU: Intel Core I5
go version: 1.15.6
clang version: 12.0.0
llvm source info : 10.0.1
tinygo source version is the last from GitHub.

"make llvm-build" OK, but "make" happened a error;

error info : 
# runtime/cgo
ld: library not found for -llibxml2.tbd
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [tinygo] Error 2

kenbell, 2021-01-30:
I can confirm this bug trying to build llvm from source.

I don't fully understand why it's broken, but as far as I can tell this is an issue with LLVM that got fixed by reverting a change sometime in aug/sept 2020 (https://github.com/ziglang/zig/issues/6087)

aykevl, 2021-02-06:
Is this still an issue?
Note that TinyGo now uses LLVM 11, so if the llvm-source directory is still on LLVM 10 you will need to remove (or rename) the `llvm-project` and `llvm-build` directories and run `make llvm-build`.

kenbell, 2021-02-07:
Looks like it's still a problem...
```
% uname -a
Darwin XXXXXX 20.3.0 Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64 x86_64

% rm -rf llvm-project 

% rm -rf llvm-build

% make llvm-build
git clone -b xtensa_release_11.0.0 --depth=1 https://github.com/tinygo-org/llvm-project llvm-project
Cloning into 'llvm-project'...

… snip …

% make
CGO_CPPFLAGS="-I/Users/kennethbell/work/tinygo/llvm-project/llvm/include -I/Users/kennethbell/work/tinygo/llvm-build/include -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/Users/kennethbell/work/tinygo/llvm-build/tools/clang/include -I/Users/kennethbell/work/tinygo/llvm-project/clang/include -I/Users/kennethbell/work/tinygo/llvm-project/lld/include" CGO_CXXFLAGS="-std=c++14" CGO_LDFLAGS="/Users/kennethbell/work/tinygo/llvm-build/lib/libclang.a -L/Users/kennethbell/work/tinygo/llvm-build/lib  -lclangAnalysis -lclangARCMigrate -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangStaticAnalyzerCheckers -lclangStaticAnalyzerCore -lclangStaticAnalyzerFrontend -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions  -lstdc++  -llldCOFF -llldCommon -llldCore -llldDriver -llldELF -llldMachO -llldMinGW -llldReaderWriter -llldWasm -llldYAML  -L/Users/kennethbell/work/tinygo/llvm-build/lib -Wl,-search_paths_first -Wl,-headerpad_max_install_names -lLLVMOption -lLLVMMCJIT -lLLVMLTO -lLLVMPasses -lLLVMObjCARCOpts -lLLVMExtensions -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMCoverage -lLLVMCoroutines -lLLVMipo -lLLVMInstrumentation -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMFrontendOpenMP -lLLVMAsmParser -lLLVMXtensaDisassembler -lLLVMXtensaCodeGen -lLLVMXtensaAsmParser -lLLVMXtensaDesc -lLLVMXtensaInfo -lLLVMAVRDisassembler -lLLVMAVRCodeGen -lLLVMAVRAsmParser -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyInfo -lLLVMRISCVDisassembler -lLLVMRISCVCodeGen -lLLVMRISCVAsmParser -lLLVMRISCVDesc -lLLVMRISCVUtils -lLLVMRISCVInfo -lLLVMAArch64Disassembler -lLLVMAArch64CodeGen -lLLVMAArch64AsmParser -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMARMDisassembler -lLLVMARMCodeGen -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMCFGuard -lLLVMGlobalISel -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMDebugInfoDWARF -lLLVMCodeGen -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMObject -lLLVMTextAPI -lLLVMMCParser -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMX86Desc -lLLVMMCDisassembler -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMDebugInfoMSF -lLLVMBinaryFormat -lLLVMX86Info -lLLVMSupport -lLLVMDemangle -lm -llibxml2.tbd -lstdc++ " go build -buildmode exe -o build/tinygo -tags byollvm -ldflags="-X main.gitSha1=`git rev-parse --short HEAD`" .
# runtime/cgo
ld: library not found for -llibxml2.tbd
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [tinygo] Error 2
```

tobiaskohlbau, 2021-02-07:
I've ran into the same issue. Looks like the fix was reverted in https://reviews.llvm.org/rGcdcb9ab10e53ff08293915af3cd897c42112bcc5.

Citing from https://bugs.llvm.org/show_bug.cgi?id=47226:
>Sounds like the patch mentioned above was backed out.
>
>I'm hesitant to block the release on this since it's not a recent regression, and the fixes for these things seem to break just >as much as they fix, so merging them so late in the release process seems like a bad idea.

So by the looks of it there is no fix included in 11 upstream. I've worked around that issue by modifying the makefile to replace _libxml2.tbd_ with _xml2_.

```
diff --git a/Makefile b/Makefile
index 69eacad..bc0efd1 100644
--- a/Makefile
+++ b/Makefile
@@ -79,10 +79,9 @@ LLD_LIBS = $(START_GROUP) -llldCOFF -llldCommon -llldCore -llldDriver -llldELF -
 ifneq ("$(wildcard $(LLVM_BUILDDIR)/bin/llvm-config*)","")
     CGO_CPPFLAGS+=$(shell $(LLVM_BUILDDIR)/bin/llvm-config --cppflags) -I$(abspath $(LLVM_BUILDDIR))/tools/clang/include -I$(abspath $(CLANG_SRC))/include -I$(abspath $(LLD_SRC))/include
     CGO_CXXFLAGS=-std=c++14
-    CGO_LDFLAGS+=$(LIBCLANG_PATH) -L$(abspath $(LLVM_BUILDDIR)/lib) $(CLANG_LIBS) $(LLD_LIBS) $(shell $(LLVM_BUILDDIR)/bin/llvm-config --ldflags --libs --system-libs $(LLVM_COMPONENTS)) -lstdc++ $(CGO_LDFLAGS_EXTRA)
+    CGO_LDFLAGS+=$(LIBCLANG_PATH) -L$(abspath $(LLVM_BUILDDIR)/lib) $(CLANG_LIBS) $(LLD_LIBS) $(subst libxml2.tbd,xml2,$(shell $(LLVM_BUILDDIR)/bin/llvm-config --ldflags --libs --system-libs $(LLVM_COMPONENTS))) -lstdc++ $(CGO_LDFLAGS_EXTRA)
 endif

-
 clean:
        @rm -rf build
```

deadprogram, 2021-02-21:
Looks like this has been fixed by #1645 so marking for next release.

tobiaskohlbau, 2021-02-23:
Sounds like the best option. I wasn't aware that _tinygo_ does not depend on _libxml_. By any means I assumed it was needed 😄

tobiaskohlbau, 2021-02-24:
I finally found some time to checkout the latest dev branch and can confirm that #1645 fixed this issue.

```
tobias@Ampere build % ./tinygo version
tinygo version 0.17.0-dev-afcc9d6 darwin/arm64 (using go version go1.16beta1 and LLVM version 11.0.0)
```

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram