tinygo-org/tinygo #1580: Arduino flash issues

pierreis, 2021-01-16:
On a Mac, I continuously get issues when flashing an Arduino Uno board via tinygo, with the usual error:

```
playground » tinygo flash -target arduino -port=$NANO33_DEV_PATH -scheduler=coroutines ./main.go                                                                            ~/L/P/a/playground

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/var/folders/rc/3gb9kzw52_l1bdyfb9pr9lh00000gn/T/tinygo969940853/main.hex"
avrdude: writing flash (6224 bytes):

Writing | ######                                             | 12% 0.12savrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
Writing | #######                                            | 14% 11.20savrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
[continued forever]
```

Note that the flashing sometimes works fine (but mostly doesn't).

I do not have any issues whatsoever flashing the board using the Arduino IDE ; only with `tinygo` from the command line.

pierreis, 2021-01-16:
Thanks to some help on discord, I have tested using `arduino-cli` for flashing, which consistently works just fine:

```
tinygo build -scheduler=tasks -target=arduino -o /tmp/test.hex ./main.go && arduino-cli upload -b arduino:avr:uno -p $UNO_DEV_PATH -i /tmp/test.hex .
```

deadprogram, 2021-01-24:
Perhaps you have an older version of avr-dude installed? That is what the `tinygo flash` command currently uses "under the hood". Did you install avr-dude via homebrew?

pierreis, 2021-01-24:
I did install via homebrew indeed.
Installed version is 6.3, which seems to be the latest.

deadprogram, 2021-01-24:
Which macOS version?

pierreis, 2021-01-24:
11.1

deadprogram, 2021-01-24:
I am still running Catalina on my dev machine. I wonder if that is related?

pierreis, 2021-01-24:
I suppose that is a possibility. Anything I could provide to help debug this?

aykevl, 2023-10-17:
Oops, sorry I never got to this issue!
I suspect this can be solved using `-target=arduino-nano-new`.

peritonow, 2024-04-26:
indeed it does. thank you

peritonow, 2024-04-26:
this issue can be close ...

closed 2024-04-26 by deadprogram