tinygo-org/tinygo #1586: panic: runtime error: goroutine stack overflow on feather-nrf52840

antonfisher, 2021-01-21:
Hi!

I'm trying TinyGo on Feather nRF52840, and I've noticed that after sometime controller stops working.
When I connect it to the USB to read serial port output (`screen /dev/tty.usbmodem141101 19200`), I get this:
```
panic: runtime error: goroutine stack overflow
```

This is my main.go:
```go
package main

import (
	"time"
)

func main() {
	go print1()
	go print2()

	for {
		println("working...")
		time.Sleep(time.Millisecond * 1000)
	}
}

func print1() {
	i := 0
	for {
		time.Sleep(time.Millisecond * 1000)
		println("print1,", i)
		i++
	}
}

func print2() {
	time.Sleep(time.Millisecond * 500)
	i := 0
	for {
		time.Sleep(time.Millisecond * 1000)
		println("print2,", i)
		i++
	}
}
```

I flash it with this command:
```shell
tinygo flash -target=feather-nrf52840 main.go
```

And after that, I connect to the serial port. In a short period of time, the program panics:
```
working...
print1, 2
print2, 2
working...
print1, 3
print2, 3
panic: runtime error: goroutine stack overflow
```

Notes:
- first time I noticed it was when I had only a single goroutine in my code, and the controller stopped working after about 2-3 hours
- it happens with one or more goroutines
- it is easy to repro if I'm connected to the serial port -- that speeds up the panic.

I found `-scheduler=coroutines` workaround here: https://github.com/tinygo-org/tinygo/issues/1009#issuecomment-605680032, and it works. I wonder if I'm doing something wrong or can help to make it work with the default scheduler.

Thanks for the great project!

deadprogram, 2021-04-22:
Is this problem still occurring with the latest TinyGo?

deusnefum, 2021-04-23:
I am running into a very similar problem with an itsybitsy-M4. I'm using goroutines that loop and sleep and eventually (but not always) I get a panic: runtime error: g (it cuts off here, but I think this is the only runtime panic error that starts with a g).

I'll try switching scheduler and report back.

tinygo version 0.17.0 linux/amd64 (using go version go1.16.3 and LLVM version 11.1.0)

aykevl, 2021-04-23:
I think I've found the issue: the FPU is somehow enabled. Not sure how it got enabled, the FPU is currently unsupported by TinyGo (it will result in a stack overflow as you've all seen now).

This also shows there is a bug in the NXP code, here:

https://github.com/tinygo-org/tinygo/blob/80caf2dab2523d1d6ff2a0716fd5fc3c83821074/src/runtime/runtime_nxpmk66f18.go#L78

With the FPU enabled, interrupts require a bigger stack to store all floating point registers which is not taken into account when calculating the stack size.

aykevl, 2021-04-23:
I _suspect_ it's getting enabled by the UF2 bootloader. So that gives two options:

1. Easy: disable the FPU at startup to avoid these stack overflows.
2. Harder: fully support the FPU, including in interrupt stack calculations.

If you want a quick fix, you can do the first thing in your code with this line:

```go
(*volatile.Register32)(unsafe.Pointer(uintptr(0xE000ED88))).Set(0)
```

Floating point operations are still supported just like before because we've been using softfloat all along. The difference is that the (unused) FPU is disabled.

deusnefum, 2021-04-23:
I hadn't realized TinyGo wasn't using the FPU. Goes to show how powerful the M4 and Go are together. I'm making extensive use of floats in my code--doing softfloat on other MCU platforms drove my code to a stall. I think it'd be great eventually can support the FPU. I'll just disable it for now.

deusnefum, 2021-04-23:
Switching to coroutines scheduler fixed my issues of the panic goroutine stackoverflow and also let me use both a pin-interrupt and i2c at the same time. If I try to use a pin-interrupt and i2c with tasks scheduler, i2c just straight doesn't work.

aykevl, 2021-04-24:
@antonfisher @deusnefum can you try this PR and see whether the stack overflows still occur? https://github.com/tinygo-org/tinygo/pull/1836

> Switching to coroutines scheduler fixed my issues of the panic goroutine stackoverflow and also let me use both a pin-interrupt and i2c at the same time.

Note that the coroutines scheduler may not always interact well with interrupts. It's best to use `-scheduler=tasks` (the default) if possible.
Additionally, it sounds like you're hitting a different kind of bug. There is no good reason why pin interrupts and i2c couldn't work together.

deusnefum, 2021-04-24:
@aykevl, Not sure if I'm doing something wrong here, but after checking out hte disable-fpu branch and building tinygo, I get a bunch of these kinds of errors:

/home/zashi/tmp/tinygo/src/machine/machine_atsamd51.go:426:32: EIC_CTRLA_ENABLE not declared by package sam

deusnefum, 2021-04-24:
Copying over the src/ from disable-fpu branch into my system's tinygo/src directory gets me a different set of errors:

```
2021-24-04 08:38:52 [omnivore-fw:main]
zashi@athyra> cat test.go
package main

func main(){
}
2021-24-04 08:38:54 [omnivore-fw:main]
zashi@athyra> go run test.go

2021-24-04 08:38:57 [omnivore-fw:main]
zashi@athyra> tinygo run test.go
panic: unknown type: func() [recovered]
        panic: unknown type: func()

goroutine 1 [running]:
go/types.(*Checker).handleBailout(0xc000520360, 0xc0007af318)
        /usr/lib/go/src/go/types/check.go:245 +0x9f
panic(0x789440, 0xc0004251e0)
        /usr/lib/go/src/runtime/panic.go:965 +0x1b9
github.com/tinygo-org/tinygo/compiler.(*stdSizes).Sizeof(0xc000018378, 0x864fe8, 0xc00018ca50, 0x10)
        /build/tinygo/src/tinygo/compiler/sizes.go:154 +0x6bf
go/types.(*Config).sizeof(0xc00050f6c0, 0x864fe8, 0xc00018ca50, 0x0)
        /usr/lib/go/src/go/types/sizes.go:251 +0x51
go/types.(*Checker).builtin(0xc000520360, 0xc0002a1f80, 0xc0003bdb40, 0x11, 0xc0007ad000)
        /usr/lib/go/src/go/types/builtins.go:611 +0x1085
go/types.(*Checker).call(0xc000520360, 0xc0002a1f80, 0xc0003bdb40, 0x1)
        /usr/lib/go/src/go/types/call.go:51 +0x828
go/types.(*Checker).exprInternal(0xc000520360, 0xc0002a1f80, 0x867368, 0xc0003bdb40, 0x0, 0x0, 0x203000)
        /usr/lib/go/src/go/types/expr.go:1519 +0x1e2c
go/types.(*Checker).rawExpr(0xc000520360, 0xc0002a1f80, 0x867368, 0xc0003bdb40, 0x0, 0x0, 0x40)
        /usr/lib/go/src/go/types/expr.go:1029 +0x85
go/types.(*Checker).multiExpr(0xc000520360, 0xc0002a1f80, 0x867368, 0xc0003bdb40)
        /usr/lib/go/src/go/types/expr.go:1652 +0x58
go/types.(*Checker).initVars.func1(0xc0002a1f80, 0x0)
        /usr/lib/go/src/go/types/assignments.go:226 +0x5d
go/types.unpack(0xc0007ad970, 0x1, 0xc0002a1e00, 0x7ff59c, 0x10, 0xc0003ab770)
        /usr/lib/go/src/go/types/call.go:193 +0x66
go/types.(*Checker).initVars(0xc000520360, 0xc0005103e0, 0x1, 0x1, 0xc0003adda0, 0x1, 0x1, 0xbf5c)
        /usr/lib/go/src/go/types/assignments.go:226 +0xe5
go/types.(*Checker).stmt(0xc000520360, 0x5, 0x867878, 0xc0003c42c0)
        /usr/lib/go/src/go/types/stmt.go:457 +0x3b18
go/types.(*Checker).stmtList(0xc000520360, 0x5, 0xc0003c42e0, 0x2, 0x2)
        /usr/lib/go/src/go/types/stmt.go:120 +0xd1
go/types.(*Checker).stmt(0xc000520360, 0x0, 0x8679c8, 0xc0003c0f60)
        /usr/lib/go/src/go/types/stmt.go:561 +0x2485
go/types.(*Checker).stmtList(0xc000520360, 0x0, 0xc0003adf40, 0x1, 0x1)
        /usr/lib/go/src/go/types/stmt.go:120 +0xd1
go/types.(*Checker).funcBody(0xc000520360, 0xc0002f0ba0, 0xc00037f310, 0x4, 0xc00051faa0, 0xc0003c0f90, 0x0, 0x0)
        /usr/lib/go/src/go/types/stmt.go:42 +0x1f3
go/types.(*Checker).funcDecl.func1()
        /usr/lib/go/src/go/types/decl.go:749 +0x67
go/types.(*Checker).processDelayed(0xc000520360, 0x0)
        /usr/lib/go/src/go/types/check.go:291 +0x3e
go/types.(*Checker).checkFiles(0xc000520360, 0xc0003d1300, 0x6, 0x8, 0x0, 0x0)
        /usr/lib/go/src/go/types/check.go:267 +0xd0
go/types.(*Checker).Files(...)
        /usr/lib/go/src/go/types/check.go:250
go/types.(*Config).Check(0xc00050f6c0, 0xc000208457, 0x7, 0xc00007edc0, 0xc0003d1300, 0x6, 0x8, 0xc0002308b8, 0xc000277ef0, 0x0, ...)
        /usr/lib/go/src/go/types/api.go:370 +0x151
github.com/tinygo-org/tinygo/loader.(*Package).Check(0xc0002307e0, 0x0, 0x0)
        /build/tinygo/src/tinygo/loader/loader.go:349 +0x1a5
github.com/tinygo-org/tinygo/loader.(*Program).Parse(0xc0002a6160, 0xc000187948, 0x1)
        /build/tinygo/src/tinygo/loader/loader.go:274 +0xd9
github.com/tinygo-org/tinygo/builder.Build(0x7ffd23482cbe, 0x7, 0x7f93fc, 0x4, 0xc0002b20c0, 0xc0007afb90, 0x0, 0x0)
        /build/tinygo/src/tinygo/builder/build.go:80 +0x5da
main.Run(0x7ffd23482cbe, 0x7, 0xc00029e000, 0xe, 0x0)
        /build/tinygo/src/tinygo/main.go:519 +0xb2
main.main()
        /build/tinygo/src/tinygo/main.go:986 +0x215b

```

aykevl, 2021-04-24:
> /home/zashi/tmp/tinygo/src/machine/machine_atsamd51.go:426:32: EIC_CTRLA_ENABLE not declared by package sam

You probably need to run `make gen-device`.

> Copying over the src/ from disable-fpu branch into my system's tinygo/src directory gets me a different set of errors:

Are you using the src directory of the dev branch to the release version of TinyGo? That's not at all supported and I'm not surprised it crashes. The TinyGo binary always has to match the src directory it's using.

deusnefum, 2021-04-24:
make gen-device was indeed the step I was missing.  Sorry for the other noise, I'm still very new to tinygo.

I think I got this right, now. I did a make build/release and used the binary in build and TINYGOROOT created in build/release/tinygo

TINYGOROOT=/home/zashi/tmp/tinygo/build/release/tinygo/ ~/tmp/tinygo/build/tinygo build -target=itsybitsy-m4 -o fw.uf2 main.go

It certainly works better now, but not perfectly. My I2C magnetometer (lis2mdl) will connect and display values without issue while a pin-interrupt is configured for all 33 seconds before the panic.

First run I got some output for less than a minute before getting: fatal error: data bus error (precise) with bus fault address 0x61000000 with sp=0x2000

And the next two runs after about 33 seconds I get the good ol panic: runtime error: g

antonfisher, 2021-05-14:
@aykevl @deadprogram, just flashed the example using `tinygo version 0.18.0 darwin/amd64 (using go version go1.16.4 and LLVM version 11.0.0)`:
```bash
tinygo flash -target=feather-nrf52840 main.go
```
And everything is working fine! Thank you!

(flashing with ```-scheduler=coroutines``` works as well 👍)

deadprogram, 2021-05-15:
That is good news @antonfisher thanks for testing. Now closing this issue, please reopen if needed.

closed 2021-05-15 by deadprogram