tinygo-org/tinygo #1587: Raspberry Pi RP2040

alankrantas, 2021-01-21:
Something wild just appears:

https://www.raspberrypi.org/products/raspberry-pi-pico/
https://www.raspberrypi.org/products/raspberry-pi-pico/specifications/

Dual-core Cortex-M0+ (up to 133 MHz) with UF2 bootloader for 4$. Both Adafruit and Arduino will have their own RP2040 version as well.

quaintdev, 2021-01-21:
This is going to be game changer for tinygo. 

Last time I tried to use tinygo with Arduino and ESP01 I was getting [RAM overflow](https://github.com/tinygo-org/drivers/issues/221). Pretty sure that issue will be resolved with RP2040.

deadprogram, 2021-01-21:
Also see https://github.com/posborne/cmsis-svd/pull/123

aykevl, 2021-01-25:
For anybody wanting to implement support, please take a look here: https://twitter.com/therealjpster/status/1353063981619965956?s=19
It contains important information on the reset sequence.

rajivmod143, 2021-02-09:
Just curious to know if anyone's working on this?

utahcon, 2021-02-09:
> Just curious to know if anyone's working on this?

> Also see [posborne/cmsis-svd#123](https://github.com/posborne/cmsis-svd/pull/123)

Nerzal, 2021-02-16:
I'm happy to help out at this topic, as soon as my pi picos arrive. Ordered 4, but they are currently not deliverable.

deadprogram, 2021-02-16:
https://www.cnx-software.com/2021/01/27/a-closer-look-at-raspberry-pi-rp2040-programmable-ios-pio/ looks helpful.

jeffmhastings, 2021-02-17:
I've got the blinky1 example running on a raspberry pi pico. Right now the sleep timing is a bit off, and there are a number of things I still need to clean up before I would be ready to submit a PR. I hope to have one ready in the next couple of days though.

sakthimaan, 2021-03-06:
any progress on this..?

kurtmc, 2021-04-10:
I just received a couple of these boards and I am keen to help out getting tinygo working. I am not really familiar with the process of adding new boards but I am happy to checkout forks and test them.

@jeffmhastings I have checked out your branch and managed to get the blinky example going, nice work!

I have noticed that `putchar` is not implemented yet (https://github.com/jeffmhastings/tinygo/blob/rp2040/src/runtime/runtime_rp2040.go#L144) I assume this is important for getting the `println` function to work. I had a quick go at implementing it but I am pretty new to micro controllers and it would probably be easier for someone how knows what they are doing :joy:

kenbell, 2021-04-11:
I've got a couple (pico and an Adafruit RP2040 feather) - will take a look if no-one gets there first.  Waiting on FedEx to deliver me a J-Link though.  In my experience SWD and a logic analyzer makes this stuff so much easier.

deadprogram, 2021-04-22:
Is anyone actively working on this? :crossed_fingers:

rythmkraze, 2021-04-23:
Ive been working on a minimal integration, just enough to run the blinky1 example, based on the Pico C SDK.

deadprogram, 2021-04-23:
@rythmkraze I assume you have seen the @jeffmhastings fork?

rythmkraze, 2021-04-24:
@deadprogram Yes I have.

bartekpacia, 2021-05-15:
I second @lexfrei, where can we expect more info?😄

soypat, 2021-05-16:
As far as I know there are two attempts at bringing RP2040 support to TinyGo. Both seem abandoned 
* https://github.com/jeffmhastings/tinygo/tree/rp2040 (last commit February)
* https://github.com/rythmkraze/tinygo/tree/pico (last commit April)

To anyone else who embarks on this apparently perilous journey, **godspeed**.

rythmkraze, 2021-05-16:
branch [pico](https://github.com/rythmkraze/tinygo/tree/pico)  has been updated. 
Contains minimal implementation to get blinky1 example working. Only supports the machine.Pin API and that too in PinOutput mode.

lexfrei, 2021-05-17:
@rythmkraze is there any ETA? No pressure, just an interest.

aykevl, 2021-05-19:
@rythmkraze can you please make a PR with that branch? That makes it a lot easier to review the changes for eventual inclusion.

a-h, 2021-07-03:
I noticed that the latest release has support, and it was added to the README. 

https://github.com/tinygo-org/tinygo/releases/tag/v0.19.0

Is this... closed?

deadprogram, 2021-07-07:
Yes, I think we can close this issue now. Thank you everyone who helped get this added!

closed 2021-07-07 by deadprogram