tinygo-org/tinygo #1591: Interface assert failed

aykevl, 2021-01-22:
The following code works with the Go compiler but fails to run in TinyGo:

```go
package main

import (
	"reflect"
)

type T int

func (t T) Foo() int {
	return 42
}

type I interface {
	Foo() int
}

func main() {
	v := reflect.ValueOf(struct{
		X T
	}{})
	x := v.Field(0).Interface().(I)
	println("Foo:", x.Foo())
}
```

Runnable sample: https://play.golang.org/p/_KB8btFIcan

Go:

```
Foo: 42
```

TinyGo:

```
panic: runtime error: type assert failed
```

I have not yet investigated why it fails.

aykevl, 2021-01-28:
I know what causes this, but it's not trivial to fix.

dgryski, 2023-01-31:
This appears to be fixed:
```
~/go/src/github.com/dgryski/bug $ go run bug.go
Foo: 42
~/go/src/github.com/dgryski/bug $ tinygo run bug.go
Foo: 42
~/go/src/github.com/dgryski/bug $ tinygo run -target=wasi bug.go
Foo: 42
```

aykevl, 2023-02-02:
Thanks for confirming, closing now! It was probably fixed quite a while ago.

closed 2023-02-02 by aykevl