tinygo-org/tinygo #1597: Undefined reference to divmoddi4 in AVR

Nerzal, 2021-01-25:
TinyGo Version:

```bash
tinygo version 0.16.0 linux/amd64 (using go version go1.15.2 and LLVM version 10.0.1)
```
TinyGo env:

```
tinygo env
GOOS="linux"
GOARCH="amd64"
GOROOT="/usr/local/go"
GOPATH="/home/tobias/go:/home/tobias/go/src/github.com/tinygo-org/tinygo"
GOCACHE="/home/tobias/.cache/tinygo"
CGO_ENABLED="1"
TINYGOROOT="/usr/local/lib/tinygo"
```

I just came across a when using float32 on Arduino UNOs.
It feels like this could be a similar problem as #1415

I bet i can easily work around it by not using any floats at all

The code:  https://gist.github.com/Nerzal/dd78e7df00eb6084dd3f3de29e37a3a8

The error i got is: 

```
tinygo flash --target=arduino supersonic-distance-sensor-example/main.go
/tmp/tinygo000479414/main.o: In function `LBB23_28':
main.go:(.text.time.Since+0x554): undefined reference to `__divmoddi4'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf-Fehler: Gefundene Dwarf-Version ist »0«, es können aber nur Informationen der Version 2, 3 und 4 gelesen werden.
/tmp/tinygo000479414/main.o: In function `LBB24_12':
main.go:(.text.command-line-arguments.main+0x190): undefined reference to `__divmoddi4'
collect2: Fehler: ld gab 1 als Ende-Status zurück
error: failed to link /tmp/tinygo000479414/main: exit status 1
``` 

This:
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf-Fehler: Gefundene Dwarf-Version ist »0«, es können aber nur Informationen der Version 2, 3 und 4 gelesen werden.

Roughly translates to: 
Dwarf-Error: Found Dwarf-Version is >>0>>, but only information from version 2,3 and 4 can be read.

aykevl, 2021-01-26:
Both of these issues should be fixed in LLVM 11, can you try with TinyGo compiled with LLVM 11?

(You may want to use `-tags=llvm11` if you compile from source).

Nerzal, 2021-01-26:
Sadly this also results in the same error

```/tmp/tinygo316342443/main.o: In function `LBB22_33':
main.go:(.text.command-line-arguments.main+0x6cc): undefined reference to `__divmoddi4'
```

Nerzal, 2021-01-26:
I can confirm, that this issue is solved using llvm11

aykevl, 2021-01-28:
Closing as this issue is resolved.

closed 2021-01-28 by aykevl

bxparks, 2023-03-10:
In case anyone else gets a similar error message with the latest tinygo version 0.27.0, the solution was to blow the cache with:
```
$ tinygo clean
```

Error message:
```
$ tinygo version
tinygo version 0.27.0 linux/amd64 (using go version go1.19.3 and LLVM version 15.0.0)

$ tinygo build -target=arduino-nano
tinygo:ld.lld: error: undefined symbol: __udivmodhi4
>>> referenced by gc_blocks.go:246 (/usr/local/lib/tinygo/src/runtime/gc_blocks.go:246)
>>>               /home/brian/.cache/tinygo/thinlto/llvmcache-D10F6DA2E6055022ACF8636F845ED740C41E16F2:(runtime.run)
>>> did you mean: __udivmodsi4
>>> defined in: /home/brian/.cache/tinygo/thinlto/llvmcache-D10F6DA2E6055022ACF8636F845ED740C41E16F2
failed to run tool: ld.lld
error: failed to link /tmp/tinygo3061748889/main: exit status 1