tinygo-org/tinygo #1598: Teensy 4.1 support

AllAwesome497, 2021-01-26:
Is there any plans in place for supporting the teensy 4.1?

From what I can tell, it uses the same chip as the teensy 4.0, so I think a good bit of it can is done in #1425.

I'm happy to help with it, though I'm not sure what to do with most of it so I don't know if I'll be able to help much.

Thanks.

ardnew, 2021-01-31:
Most of the Teensy 4.0 support can indeed be re-used for Teensy 4.1 - the entire `runtime` package in particular.

To add Teensy 4.1 support, a new board file `src/machine/board_teensy41.go` defining how pins are broken out on the PCB, target JSON `targets/teensy41.json` and linker script `targets/mimxrt1062-teensy41.ld`/`targets/teesny41.s` need to be added. You can probably re-use the Teensy 4.0 as template for most of this.

See the Wiki page for general concepts on what is required: https://github.com/tinygo-org/tinygo/wiki/Adding-a-new-board

AllAwesome497, 2021-02-01:
Oh yep, I've actually been doing a little bit of work on this. The main issue that i have is not knowing if i did things the correct way. I also can't get serial communication to work which is annoying.

AllAwesome497, 2021-02-01:
All i've been able to do is get it to turn a light on or off at the start, and i haven't been able to get any timing to work and as such can't tell if im able to turn a light on and off

ardnew, 2021-02-02:
If you want to post a link to your branch I'd be happy to review

AllAwesome497, 2021-02-07:
@ardnew so a bit of an update: 
Here's what I've done:
- `board_teensy41.go` (copied board_teensy40.go)
-  `mimxrt1062-teensy41.ld` (copied `mimxrt1062-teensy40.ld` just changed the memory area here to match the ld file in the cores repo for tinygo)
- `teensy41.s` (copied `teensy40.s`, changed the `sflashA1Size` to the value that's commented out)
- `teensy41.json` (Copied `teensy40.json` just updated to use the correct build flags and point to the new assembly and linkerscript files)

Problem is, I can't seem to get anything working (no uart, no gpio). The only indication that its even being flashed is that it overrides the code that I flash with arduino.

uart also isn't working over usb, no `/dev/ttyACM*`. I also made an arduino program with my arduino that would read from the pins directly and echo over usb which worked w/ serial in arduino code but not w/ tinygo.

AllAwesome497, 2021-02-09:
I've got a branch of a fork with what I've done so far if you want to take a look: https://github.com/AllAwesome497/tinygo/tree/board-teensy41.

davecheney, 2026-10-24:
Supported with `-target=teensy41`, including SPI and I2C. USB support was added in https://github.com/tinygo-org/tinygo/pull/5691, though Teensy 4.1 hardware testing remains unconfirmed there. Please open separate issues for any remaining problems.

closed 2026-10-24 by davecheney