tinygo-org/tinygo #1599: Build failures from removing the callMain workaround

deadprogram, 2021-01-26:
PR #1592 had an unexpected side effect that is causing a few of the builds in the drivers repo to fail. I have confirmed that this is the commit that introduced the failure. Here is an example:

```
$ tinygo build -size short -o ./build/test.hex -target=itsybitsy-m0 ./examples/hcsr04/main.go
tinygo:ld.lld: error: undefined symbol: main.main
>>> referenced by scheduler_any.go:21 (/home/ron/Development/tinygo/tinygo/src/runtime/scheduler_any.go:21)
>>>               /tmp/tinygo581397949/main.o:(runtime.run$1$gowrapper)
failed to run tool: ld.lld
error: failed to link /tmp/tinygo581397949/main: exit status 1
```

The `hcsr04`, `ssd1331`, and `ssd1351` drivers are the three that fail with the same error as above.

aykevl, 2021-01-27:
Found the issue, it's not in the compiler but in the drivers repo. The compiler previously accepted invalid code.
See: https://github.com/tinygo-org/drivers/pull/224

aykevl, 2021-01-28:
It looks like this issue is now fixed?

deadprogram, 2021-01-28:
Yes. Now closing.

closed 2021-01-28 by deadprogram