tinygo-org/tinygo #1602: Cannot compile on linux when import "net"

RudolfVonKrugstein, 2021-01-29:
When trying to compile this:

```
package main

import (
	"fmt"
	_ "net"
)

func main() {
	fmt.Println("Hello")
}
```

with these comands:

```
# docker run --privileged --rm -v $(pwd):/src -ti tinygo/tinygo bash
> cd /src
> tinygo run main.go
```

I get:

```
# net
../usr/include/netdb.h:617:23: unexpected token ILLEGAL
../usr/include/netdb.h:618:22: unexpected token ILLEGAL
../usr/include/netdb.h:624:23: unexpected token ILLEGAL
../usr/include/netdb.h:625:25: unexpected token ILLEGAL
../usr/include/netdb.h:617:23: unexpected token ILLEGAL
../usr/include/netdb.h:618:22: unexpected token ILLEGAL
../usr/include/netdb.h:624:23: unexpected token ILLEGAL
../usr/include/netdb.h:625:25: unexpected token ILLEGAL
../usr/include/netdb.h:617:23: unexpected token ILLEGAL
../usr/include/netdb.h:618:22: unexpected token ILLEGAL
../usr/include/netdb.h:624:23: unexpected token ILLEGAL
../usr/include/netdb.h:625:25: unexpected token ILLEGAL
```

I get the same error, when I have included the bluetooth package (https://github.com/tinygo-org/bluetooth).

So I can not compile for linux with the bluetooth package.

rzr, 2021-03-09:
net API is not yet supported according to:

 https://tinygo.org/lang-support/stdlib/#net

    # net
    /home/ron/.gvm/gos/go1.15/src/net/parse.go:80:12: st.ModTime undefined (type os.FileInfo has no field or method ModTime)

Relate-to: https://github.com/rzr/webthing-go/issues/4

davecheney, 2026-10-26:
This import-only example now compiles with TinyGo 0.44.0-dev-eae3f875 for linux/amd64:

```go
package main

import (
	"fmt"
	_ "net"
)

func main() {
	fmt.Println("Hello")
}
```

```sh
GOOS=linux GOARCH=amd64 tinygo build -o main main.go
```

Verified by cross-compiling on darwin/arm64. This checks the reported net import failure, not networking or Bluetooth at runtime. Please open a new issue if you encounter a failure with a current TinyGo version.

closed 2026-10-26 by davecheney