tinygo-org/tinygo #1619: support darwin arm64

tobiaskohlbau, 2021-02-07:
This issue is to track what is needed to support darwin arm64 on tinygo out of the box.

First of all I've got tinygo working on the new machines without much hassle. The only issue I encountered was that tinygo depends on `go.bug.st/serial v1.0.0` which has a build constraint which excludes arm64 and darwin in combination. I created an upstream PR to fix this https://github.com/bugst/go-serial/pull/96.

Tinygo could switch to my temporary fork of that repository until the PR is merged or it could wait for the completion before supporting arm64 on darwin.

deadprogram, 2021-02-07:
This is seriously awesome, thank you for working on it! I suppose we should give the maintainers at least a day to merge that PR. :smile_cat:

tobiaskohlbau, 2021-02-07:
I think it's even a good idea to give them at least time until Go 1.16 is released. As Go 1.16 would obviously also be a requirement for this to work. As far as I expect 1.16 should be right around the corner.

deadprogram, 2021-02-09:
> As far as I expect 1.16 should be right around the corner.

Yes, the next release of TinyGo will come out shortly after this next Go release.

tobiaskohlbau, 2021-02-15:
The upstream project merged the darwin/arm64 support and released version https://github.com/bugst/go-serial/tree/v1.1.2 which includes the support.

deadprogram, 2021-02-15:
@tobiaskohlbau will you submit a PR here that updates the go.mod/go.sum files?

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram