tinygo-org/tinygo #1620: Cannot get cgo to work

oflebbe, 2021-02-07:
Made a demonstration project, 
https://github.com/oflebbe/testesp32

modelled after tinygo-org/bluetooth 

Somehow the c source is not picked up. Please tell me what I am missing.

aykevl, 2021-02-07:
Your code sample is not valid:

```
$ tinygo run ./example/main.go 
example/main.go:3:8: found packages testesp32 (demo.go) and main (main.go) in /home/ayke/tmp/testesp32
package command-line-arguments

$ go run ./example/main.go
example/main.go:3:8: found packages testesp32 (demo.go) and main (main.go) in /home/ayke/tmp/testesp32
```

aykevl, 2021-02-07:
But if I remove the file main.go, I can reproduce your issue.

aykevl, 2021-02-07:
I think it is because TinyGo does not (yet) support variadic parameters in CGo. This will work:

```diff
--- a/demo.h
+++ b/demo.h
@@ -1 +1 @@
-extern int fortytwo();
+extern int fortytwo(void);
```

Note that in C, any function that has an empty parameter list is a variadic function. Therefore, you should put `void` in there unless (and this is rare) you really want to make a variadic function.

aykevl, 2021-02-07:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/1621

oflebbe, 2021-02-07:
Thanks. OMG.

closed 2021-02-07 by oflebbe