tinygo-org/tinygo #1622: ARM inline assembly help

mogenson, 2021-02-09:
Hello, I'm trying to use the inline assembly feature to call a bios function on the Game Boy Advance with the `swi` instruction. One example function is a software divide that uses registers `r0`, `r1`, and `r3` for input/output.

From the GBA documentation:
```
0x06: Div
Input:
  r0 = numerator
  r1 = denominator
Output:
  r0 = numerator / denominator
  r1 = numerator % denominator;
  r3 = abs (numerator / denominator)
```

Here's my Go function:
```
func Div(numerator, denominator int32) (result, remainder, abs int32) {
       device.AsmFull(`
           ldr r0 {numerator}
           ldr r1 {denominator}
           swi 0x06
           str r0 {result}
           str r1 {remainder}
           str r3 {abs}
        `, map[string]interface{}{
                "numerator":   numerator,
                "denominator": denominator,
                "result":      result,
                "remainder":   remainder,
                "abs":         abs,
        })
        return
}
```

Here's the compiler output:
```
<inline asm>:2:15: error: unexpected token in argument list
       ldr r0 r2
              ^
<inline asm>:3:15: error: unexpected token in argument list
       ldr r1 r3
              ^
<inline asm>:5:15: error: unexpected token in argument list
       str r0 r12
              ^
<inline asm>:6:15: error: unexpected token in argument list
       str r1 r12
              ^
<inline asm>:7:15: error: unexpected token in argument list
       str r3 r12
              ^
LLVM ERROR: Error parsing inline asm
```

Am I using the inline assembly syntax correctly? Also, given that registers `r0`, `r1`, etc. are used by the compiler for calling Go functions, is there a good way to not clobber these registers?

aykevl, 2021-02-09:
This is not correct ARM assembly syntax. I wonder where you got it from?

> Also, given that registers r0, r1, etc. are used by the compiler for calling Go functions, is there a good way to not clobber these registers?

Currently, no. So code like this is unsafe unless you save/restore all clobbered register according to the calling convention.
What are you trying to do? Dividing should already be well supported.

aykevl, 2021-02-09:
Also, you are loading from an integer (non address) value? `ldr` loads a value while `numerator` for example is an integer, not a pointer.

mogenson, 2021-02-09:
Thanks for the help. There are much more useful bios functions. This is a trivial example where I can easily test the result.

My mistake with the assembly, I forgot to take the values by address. Here's an updated example where I attempt to stash registers and restore them after the bios call.

```
func Div(numerator, denominator int32) (result, remainder, abs int32) {

	// make heap copies of all our local variables
	tmp_numerator := new(int32)
	*tmp_numerator = numerator

	tmp_denominator := new(int32)
	*tmp_denominator = denominator

	tmp_result := new(int32)
	*tmp_result = result

	tmp_remainder := new(int32)
	*tmp_remainder = remainder

	tmp_abs := new(int32)
	*tmp_abs = abs

	// stash r0, r1, and r3 register values on the heap
	tmp_r0 := new(uint32)
	tmp_r1 := new(uint32)
	tmp_r3 := new(uint32)

	device.AsmFull(`
	   str r0 {tmp_r0}
	   str r1 {tmp_r1}
	   str r3 {tmp_r3}
	`, map[string]interface{}{
		"tmp_r0": tmp_r0,
		"tmp_r1": tmp_r1,
		"tmp_r3": tmp_r3,
	})

	// call bios function 6: divide
	device.AsmFull(`
           ldr r0 {numerator}   # load value of numerator into r0
           ldr r1 {denominator} # load value of denominator into r1
           swi 0x06             # call bios function
           str r0 {result}      # store value of r0 into result
           str r1 {remainder}   # store value of r1 into remainder
           str r3 {abs}         # store value of r3 into abs
	`, map[string]interface{}{
		"numerator":   tmp_numerator,
		"denominator": tmp_denominator,
		"result":      tmp_result,
		"remainder":   tmp_remainder,
		"abs":         tmp_abs,
	})

	// restore r0, r1, and r3 register values
	device.AsmFull(`
	   ldr r0 {tmp_r0}
	   ldr r1 {tmp_r1}
	   ldr r3 {tmp_r3}
	`, map[string]interface{}{
		"tmp_r0": tmp_r0,
		"tmp_r1": tmp_r1,
		"tmp_r3": tmp_r3,
	})

	// copy values from heap back to local variables
	numerator = *tmp_numerator
	denominator = *tmp_denominator
	result = *tmp_result
	remainder = *tmp_remainder
	abs = *tmp_abs

	return
}
```

This results in the error:
```
<inline asm>:2:12: error: unexpected token in argument list
           str r0 {tmp_r0}
                  ^
<inline asm>:3:12: error: unexpected token in argument list
           str r1 {tmp_r1}
                  ^
<inline asm>:4:12: error: unexpected token in argument list
           str r3 {tmp_r3}
```

If I'm reading the [inline assembly](https://tinygo.org/compiler-internals/inline-assembly/) documentation right, `tmp_r0` should be recognized as a pointer and used as an address, correct? Also, I'm not sure what registers the function arguments and return values are stored in, so we may be copying registers multiple times.

aykevl, 2021-02-09:
This is still entirely unsafe because LLVM can pick any register (including `r0`-`r3`) to store the inline assembly for the parameters.

Can you tell me which function you are trying to call? That would make it easier to explain how to do this.

mogenson, 2021-02-10:
Sure thing. [Here's a tutorial](https://www.coranac.com/tonc/text/swi.htm) about calling bios functions using ARM assembly. Section 17.3 uses the `Div` function as an example. The inline assembly there just uses the `swi` instruction and takes advantage of the C ABI to populate the `r0` and `r1` registers.

By the way, this is just an educational interest. No worries if it's not currently possible.

aykevl, 2021-02-10:
I see.

I think the easiest way to do this using CGo. You can create a C file with the inline assembly as you'd like, and another file that calls the function in there. TinyGo doesn't have a heavy CGo overhead, a CGo call is just a regular function call.

mogenson, 2021-02-10:
Thanks, I'll look into CGo!

closed 2021-02-10 by mogenson

aykevl, 2021-02-10:
Actually, that tutorial you linked isn't safe either. If for example that `Div` function were inlined (which is likely because it is short), the C ABI would no longer apply.

I highly recommend reading this:
http://www.ethernut.de/en/documents/arm-inline-asm.html