tinygo-org/tinygo #1624: Standalone C files don't get header search path from #cgo ?

oflebbe, 2021-02-10:
Hi,

thanks for hinting for using proper prototypes.

However, only slightly more involved commands still won't compile

See https://github.com/oflebbe/testesp32 

Maybe the CFLAGS flag isn't honored for stand-alone C files ?

aykevl, 2021-02-10:
> Maybe the CFLAGS flag isn't honored for stand-alone C files ?

Yes, that's true. Fixing this shouldn't be too hard, but it conflicts with https://github.com/tinygo-org/tinygo/pull/1612 so I'd like to get that PR ready first. In the meantime, you could work around it like this:

```diff
--- a/demo.c
+++ b/demo.c
@@ -1,4 +1,4 @@
-#include <foo.h>
+#include "include/foo.h"
 
 int fortytwo()
 {
```

Basically, do an include relative to the current directory.

aykevl, 2021-02-10:
Based on the repository name, you might be interested in this PR: https://github.com/tinygo-org/tinygo/pull/1626

aykevl, 2021-02-11:
#1626 does not fix this issue, so I've removed the next-release label.

oflebbe, 2021-02-11:
Indeed !  #1626 looks very interesting.
Wanted to poke into compiling and linking with ESP-IDF Headers and binaries .

aykevl, 2021-03-12:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/1708

deadprogram, 2021-11-18:
Looks like this was addressed some time ago, so now closing. Please reopen if needed. Thank you!

closed 2021-11-18 by deadprogram