tinygo-org/tinygo #1630: I create new board,but LED not declared by package machine

wcpsoft, 2021-02-12:
Today,I use josh-mega-esp32 board witre led demo,but there was a problem,LED not declared by package machine?
base on board_esp32-coreboard-v2.go template create board_josh-mega-esp32.go, i modify file IO declared, this is my code,i want someone help me solve this problem.thank you.

i'm sure led.go use led := machine.Pin(22) run work.

- board_josh-mega-esp32.go code

`// +build josh-mega-esp32

package machine

// Built-in LED on some ESP32 boards.
const LED = IO22

const (
	CLK	Pin = 6
	CMD	Pin = 11
	IO0  Pin = 0
	IO1  Pin = 1
	IO10 Pin = 10
	IO16 Pin = 16
	IO17 Pin = 17
	IO18 Pin = 18
	IO19 Pin = 19
	IO2  Pin = 2
	IO21 Pin = 21
	IO22 Pin = 22
	IO23 Pin = 23
	IO25 Pin = 25
	IO26 Pin = 26
	IO27 Pin = 27
	IO3  Pin = 3
	IO32 Pin = 32
	IO33 Pin = 33
	IO34 Pin = 34
	IO35 Pin = 35
	IO36 Pin = 36
	IO39 Pin = 39
	IO4  Pin = 4
	IO5  Pin = 5
	IO9  Pin = 9
	RXD    Pin = 3
	SD0    Pin = 7
	SD1    Pin = 8
	SD2    Pin = 9
	SD3    Pin = 10
	SVN    Pin = 39
	SVP    Pin = 36
	TCK    Pin = 13
	TD0    Pin = 15
	TDI    Pin = 12
	TMS    Pin = 14
	TXD    Pin = 1
)

// SPI pins
const (
	SPI0_SCK_PIN = IO18
	SPI0_SDO_PIN = IO23
	SPI0_SDI_PIN = IO19
	SPI0_CS0_PIN = IO5
)

// I2C pins
const (
	SDA_PIN = IO21
	SCL_PIN = IO22
)

// ADC pins
const (
	ADC0 Pin = IO34
	ADC1 Pin = IO35
	ADC2 Pin = IO36
	ADC3 Pin = IO39
)

// UART0 pins
const (
	UART_TX_PIN = IO1
	UART_RX_PIN = IO3
)

// UART1 pins
const (
	UART1_TX_PIN = IO9
	UART1_RX_PIN = IO10
)

// PWM pins
const (
	PWM0_PIN Pin = IO2
	PWM1_PIN Pin = IO0
	PWM2_PIN Pin = IO4
)
`
- josh-mega-esp32.json code
{
	"inherits": [
		"esp32"
	],
	"build-tags": [
		"josh-mega-esp32"
	]
}

- led.go code
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 1000)
		led.High()
		time.Sleep(time.Millisecond * 1000)
	}
}

aykevl, 2021-02-12:
> ```go
> // +build josh-mega-esp32
> ```

Build tags may not contain dashes. You can try `josh_mega_esp32` instead:

> ```go
> // +build josh_mega_esp32
> ```

> ```json
> 	"build-tags": [
> 		"josh_mega_esp32"
> 	]
> ```

wcpsoft, 2021-02-13:
thanks,It has already worked.

closed 2021-02-13 by wcpsoft