tinygo-org/tinygo #1633: macOS dev build failed

hamidrezakks, 2021-02-15:
Hi,
I've been waiting so long for 0.17.0 to be released (not yet :D )
So I decided to build the dev branch myself,
Here is the error I get on `go install`:

`# github.com/tinygo-org/tinygo/cgo
cgo/libclang.go:18:10: fatal error: 'clang-c/Index.h' file not found
#include <clang-c/Index.h> // if this fails, install libclang-10-dev
         ^~~~~~~~~~~~~~~~~
1 error generated.
# tinygo.org/x/go-llvm
../../go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210206225315-7fe719483a0f/analysis.go:16:10: fatal error: 'llvm-c/Analysis.h' file not found
#include "llvm-c/Analysis.h" // If you are getting an error here read bindings/go/README.txt
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
`

I'm using go 1.15.7, llvm 11 and macOS Catalina

Bests,
KK

aykevl, 2021-02-15:
Are you using the latest version of the dev branch? Support for LLVM 11 was only merged a week ago: https://github.com/tinygo-org/tinygo/pull/1618

aykevl, 2021-02-15:
Nevermind, it looks like you are.

Can you check whether `/usr/local/opt/llvm@11` exists?

hamidrezakks, 2021-02-16:
After calling:
`llvm-gcc -v`
```
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

hamidrezakks, 2021-02-16:
And yes, directory `/usr/local/opt/llvm` exists.

aykevl, 2021-02-16:
> Apple clang version 11.0.3 (clang-1103.0.32.62)

That's a different LLVM, the one from Apple. TinyGo does not work with that variant. The path `/usr/local/opt/llvm` is also the wrong path: it has to be `/usr/local/opt/llvm@11`. At the moment only LLVM 11 from Homebrew is supported, which you can install using `brew install llvm@11`.

hamidrezakks, 2021-02-16:
Thanks, it successfully installed, but version is still 0.16.0

aykevl, 2021-02-16:
> Thanks, it successfully installed, but version is still 0.16.0

What do you mean? After `go install`, `tinygo install` still reports version 0.16.0?

hamidrezakks, 2021-02-16:
Yes

aykevl, 2021-02-17:
In that case, you are most likely not executing the right binary. Please check `which tinygo` to see which binary you are executing.

hamidrezakks, 2021-02-17:
You are right,
It was on GOROOT path

Thanks

dkegel-fastly, 2021-03-05:
I'm hitting this on mac os 10.15.7.

```brew install llvm@11```
reports
```llvm 11.1.0 is already installed and up-to-date.```

But llvm-gcc -v reports it is the apple version, despite /usr/local/opt/llvm/bin existing and being first on PATH, and clang -v reporting ```InstalledDir: /usr/local/opt/llvm@11/bin```.

Evidently ```brew install llvm``` installs llvm 11, but does not install llvm-gcc...?

aykevl, 2021-03-05:
@dkegel-fastly `llvm-gcc` is entirely unrelated to TinyGo. TinyGo needs LLVM 11 from Homebrew, which can be installed with `brew install llvm@11`.

Can you check whether the directory `/usr/local/opt/llvm@11/include` exists and contains `clang-c/Index.h`?

dkegel-fastly, 2021-03-05:
Yes:
```
Dans-MacBook-Pro:test dkegel$ ls -l /usr/local/opt/llvm@11/include/clang-c/Index.h
-rw-r--r--  1 dkegel  admin  214607 Feb  3 13:51 /usr/local/opt/llvm@11/include/clang-c/Index.h
Dans-MacBook-Pro:test dkegel$ ls -l /usr/local/opt/llvm
lrwxr-xr-x  1 dkegel  admin  21 Mar  5 09:05 /usr/local/opt/llvm -> ../Cellar/llvm/11.1.0
Dans-MacBook-Pro:test dkegel$ ls -l /usr/local/opt/llvm@11
lrwxr-xr-x  1 dkegel  admin  21 Mar  5 09:05 /usr/local/opt/llvm@11 -> ../Cellar/llvm/11.1.0
```

Evidently ```brew install llvm@11``` is now a synonym for  ```brew install llvm```.

Glad to hear I was mistaken about llvm-gcc mattering.  I misread comments earlier in this issue.

aykevl, 2021-03-05:
That's weird. Can you do some sanity checking?

* The Go version should be recent (`go version`)
* You should be using a recent version of TinyGo. In particular, cgo/libclang_config.go should have the line `#cgo darwin CFLAGS: -I/usr/local/opt/llvm@11/include`.

dkegel-fastly, 2021-03-05:
Thank you.

I followed https://tinygo.org/getting-started/macos/ but assumed I would get the development (main / master) branch.

After doing 
```
git checkout dev
git submodule update --recursive
```
all was well, and I was able to build with ```go install```.

So, another case of user error, though maybe this one could be addressed by changing the words
```Start with getting the source code:``` in https://tinygo.org/getting-started/macos/ to ```Start with getting the latest released source code:``` to give a subtle hint to the reader about what branch they'll be building.

aykevl, 2021-03-08:
> So, another case of user error, though maybe this one could be addressed by changing the words
> `Start with getting the source code:` in https://tinygo.org/getting-started/macos/ to `Start with [...]

Not sure about that one. It depends on what someone might want: the last release or the dev branch? Right now we assume people will want to build the latest release.

closed 2021-05-22 by hamidrezakks

andrewmed, 2021-08-18:
strange this is closed

on latest dev version I get the same error despite existing VERSION file
`andy@desktop frontend % ~/code/tinygo/tinygo version  
tinygo version 0.20.0-dev darwin/amd64 (using go version <unknown> and LLVM version 11.1.0)
andy@desktop frontend % ~/code/tinygo/tinygo build -o main.wasm -target wasm main.go
error: could not read version from GOROOT (/Users/andy/code/goroot): Invalid go version output:
// Code generated by go tool dist; DO NOT EDIT.

package sys

const StackGuardMultiplierDefault = 1

andy@desktop frontend % cat /Users/andy/code/goroot/VERSION                         
go1.17rc2% `

andrewmed, 2021-08-18:
Created issue https://github.com/tinygo-org/tinygo/issues/2059