tinygo-org/tinygo #1634: Package runtime/debug is not in GOROOT

bragaz, 2021-02-15:
Hi everyone, I was looking at [this](https://github.com/tinygo-org/tinygo/issues/1472) and it was closed apparently because it was related to problems in supporting wasm with older go version but I'm having the same issue with go `1.15`

```
package runtime/debug is not in GOROOT (/Users/leonardo/Library/Caches/tinygo/goroot-go1.15.5-093539b4bdf036f3692361094b6ad0277884029bf081496eed0130db628030c1/src/runtime/debug)
package command-line-arguments
        imports github.com/cosmos/cosmos-sdk/crypto/hd
        imports github.com/cosmos/cosmos-sdk/crypto/keys/secp256k1
        imports github.com/cosmos/cosmos-sdk/codec
        imports github.com/cosmos/cosmos-sdk/codec/types
```
Is it still unsupported?

yaron2, 2021-03-02:
I'm getting the same thing with Go 1.15.

aykevl, 2021-03-04:
Can you give the code that triggers it so I can reproduce the issue?

bragaz, 2021-03-05:
https://github.com/desmos-labs/go-key-manager

aykevl, 2021-03-05:
I have created a dummy package here: https://github.com/tinygo-org/tinygo/pull/1687

If you try it, you'll see that the next build failure is the lack of encoding/json support.

dgryski, 2023-01-31:
Closing.  We should file issues for the pieces from the `debug` package we actually need.

closed 2023-01-31 by dgryski