tinygo-org/tinygo #1635: crypto bigInt support

hamidrezakks, 2021-02-16:
Hi,
Is there any way to use bigInt in tinygo?

```
# crypto/rsa
/usr/local/go/src/math/big/nat.go:74:11: interp: unknown GEP

traceback:
/usr/local/go/src/math/big/nat.go:74:11:
  %16 = getelementptr inbounds i32, i32* %4, i32 0, !dbg !2064
/usr/local/go/src/math/big/nat.go:84:19:
  %9 = call { i32*, i32, i32 } @"(math/big.nat).setWord"(i32* %6, i32 %7, i32 %8, i32 %3, i8* undef, i8* undef), !dbg !2071
/usr/local/go/src/math/big/int.go:55:25:
  %17 = call { i32*, i32, i32 } @"(math/big.nat).setUint64"(i32* %14, i32 %15, i32 %16, i64 %2, i8* undef, i8* undef), !dbg !2077
/usr/local/go/src/math/big/int.go:69:26:
  %2 = call %"math/big.Int"* @"(*math/big.Int).SetInt64"(%"math/big.Int"* %0, i64 %x, i8* undef, i8* undef), !dbg !2061
crypto/rsa/<init>:38:25:
  %288 = call %"math/big.Int"* @"math/big.NewInt"(i64 0, i8* undef, i8* undef), !dbg !2189

```

hamidrezakks, 2021-02-16:
Also `fmt.Errorf` is not supported,

```
# git.ronaksoft.com/river/web-wasm/msg
/usr/local/Cellar/tinygo/0.16.0/src/sync/pool.go:14:14: interp: unsupported instruction

traceback:
/usr/local/Cellar/tinygo/0.16.0/src/sync/pool.go:14:14:
  %18 = inttoptr i32 %17 to %runtime._interface (i8*, i8*)*, !dbg !2068
/usr/local/go/src/fmt/print.go:137:17:
  %0 = call %runtime._interface @"(*sync.Pool).Get"(%sync.Pool* @fmt.ppFree, i8* undef, i8* undef), !dbg !2105
/usr/local/go/src/fmt/errors.go:18:17:
  %5 = call %fmt.pp* @fmt.newPrinter(i8* undef, i8* undef), !dbg !2114
git.ronaksoft.com/river/web-wasm/msg/<init>:5789:41:
  %12 = call %runtime._interface @fmt.Errorf(i8* getelementptr inbounds ([48 x i8], [48 x i8]* @"git.ronaksoft.com/river/web-wasm/msg.init$string.837", i32 0, i32 0), i32 48, %runtime._interface* null, i32 0, i32 0, i8* undef, i8* undef), !dbg !2062

```

I managed to fix it by replacing `fmt.Errorf` with my own function `DumpError`

```
func DumpError(err string, ...arg interface{}) {
    return errors.New(err)
}
```

aykevl, 2021-02-17:
Most of these issues should be fixed in the next release. I would suggest trying it (or waiting until we do a release, which should happen very soon).

hamidrezakks, 2021-02-17:
Thanks, I'll give it a try.

hamidrezakks, 2021-02-17:
I ended up here:

```
wasm-ld: error: cannot open /tinygo/lib/wasi-libc/sysroot/lib/wasm32-wasi/libc.a: No such file or directory
error: failed to link /var/folders/8b/f64h19s95bnctf_l89wb6qgm0000gn/T/tinygo188653196/main: exit status 1
```

aykevl, 2021-02-17:
Try running `make wasi-libc`.

hamidrezakks, 2021-02-18:
So I did this:
`export PATH=/usr/local/opt/llvm/bin:$PATH`
and then this:
`make install INSTALL_DIR=/tmp/wasi-libc`
And it worked,
Thanks

But binary output size is about 1.2MB, it is ok?

hamidrezakks, 2021-02-18:
Golang native WASM output was about 4MB

hamidrezakks, 2021-02-18:
Ok I faced another error:
TypeError: WebAssembly.instantiate(): Import #0 module="wasi_unstable" error: module is not an object or function

Seems like wasi doesn't work with go.importObject
```
WebAssembly.instantiateStreaming(response, go.importObject).then((res) => {
    console.timeEnd('wasm init');
    go.run(res.instance);
}).catch((err) => {
    console.warn(err);
});
```

aykevl, 2021-02-18:
You can try `-no-debug`, which should lower binary size. 1.2MB seems rather large.

What is the full `tinygo` command that you use?

hamidrezakks, 2021-02-20:
Build command:
`GOOS=js GOARCH=wasm /Users/hamidrezakk/go/bin/tinygo build -o /Users/hamidrezakk/ronak/river/web-app/public/bin/river-tiny.wasm -target wasm .`

But the major problem is now about `go.importObject` in `instantiateStreaming `

hamidrezakks, 2021-02-22:
I finally managed to run it, but here is the new error I get:
`WebAssembly.instantiate(): Import #5 module="env" function="math/big.addVV" error: function import requires a callable`
is math/bigInt supported.
@aykevl

closed 2021-05-22 by hamidrezakks

ramilexe, 2022-05-29:
Hi @hamidrezakks 
How did you fix issue:
> WebAssembly.instantiate(): Import #5 module="env" function="math/big.addVV" error: function import requires a callable

?