tinygo-org/tinygo #1638: build: add Go 1.16 darwin/arm64 binary build for release to CircleCI

deadprogram, 2021-02-17:
Once we have added Go 1.16 to the CircleCI build, we need to add build step(s) to produce a darwin/arm64 binary artifact for release distribution.

aykevl, 2021-02-17:
Certainly. However, I think it is most practical to wait until CircleCI has added ARM64 executors, see: https://support.circleci.com/hc/en-us/articles/360047734172-Can-I-build-Apple-Silicon-Universal-apps-

Cross compiling large C++ projects is always a hassle. Cross compiling a compiler even more so. I could spend time to try to figure out how to cross compile TinyGo, but I think that time would be better spent on something else (especially as it is a temporary workaround until CircleCI adds Apple Silicon support).

joeshaw, 2023-03-24:
CI is on GitHub Actions now, but still no `darwin/arm64` availability.  That's being tracked in https://github.com/github/roadmap/issues/528 but looks like it's not coming for a while.

HacDan, 2023-11-06:
Looks like this shipped! [🚢](https://github.blog/2023-10-02-introducing-the-new-apple-silicon-powered-m1-macos-larger-runner-for-github-actions/)

deadprogram, 2023-11-07:
Only for paying plans, not for free open source. :money_with_wings:

deadprogram, 2024-12-02:
We have had this for a while now, so closing. :smile: 

Thanks everyone!

closed 2024-12-02 by deadprogram