tinygo-org/tinygo #1643: Cannot import functions with the same name from different modules

fgsch, 2021-02-19:
Tinygo version:

`tinygo version 0.16.0 darwin/amd64 (using go version go1.15.8 and LLVM version 10.0.1)`

Example:

```
package main

//go:wasm-module module1
//export foo
func module1_foo(int32) int32

//go:wasm-module module2
//export foo
func module2_foo(int32) int32

func main() {
        module1_foo(0)
        module2_foo(0)
}
```

Command line:

`tinygo build -target wasm -o main.wasm main.go`

Output in text format (via wasm2wat):

```
(module
  (type (;0;) (func (param i32)))
  (type (;1;) (func (param i32 i32 i32 i32) (result i32)))
  (type (;2;) (func (param i32) (result i32)))
  (type (;3;) (func))
  (type (;4;) (func (param i32 i32 i32) (result i32)))
  (type (;5;) (func (param i32 i32)))
  (import "wasi_unstable" "fd_write" (func $fd_write (type 1)))
  (import "module2" "foo" (func $foo (type 2)))
  (func $__wasm_call_ctors (type 3))
  (func $memset (type 4) (param i32 i32 i32) (result i32)
    (local i32 i32 i32 i64)
    block  ;; label = @1
      local.get 2
      i32.eqz
      br_if 0 (;@1;)
      local.get 0
      local.get 1
      i32.store8
      local.get 2
      local.get 0
      i32.add
      local.tee 3
      i32.const -1
      i32.add
      local.get 1
      i32.store8
      local.get 2
      i32.const 3
      i32.lt_u
      br_if 0 (;@1;)
      local.get 0
      local.get 1
      i32.store8 offset=2
      local.get 0
      local.get 1
      i32.store8 offset=1
      local.get 3
      i32.const -3
      i32.add
      local.get 1
      i32.store8
      local.get 3
      i32.const -2
      i32.add
      local.get 1
      i32.store8
      local.get 2
      i32.const 7
      i32.lt_u
      br_if 0 (;@1;)
      local.get 0
      local.get 1
      i32.store8 offset=3
      local.get 3
      i32.const -4
      i32.add
      local.get 1
      i32.store8
      local.get 2
      i32.const 9
      i32.lt_u
      br_if 0 (;@1;)
      local.get 0
      i32.const 0
      local.get 0
      i32.sub
      i32.const 3
      i32.and
      local.tee 4
      i32.add
      local.tee 3
      local.get 1
      i32.const 255
      i32.and
      i32.const 16843009
      i32.mul
      local.tee 1
      i32.store
      local.get 3
      local.get 2
      local.get 4
      i32.sub
      i32.const -4
      i32.and
      local.tee 4
      i32.add
      local.tee 2
      i32.const -4
      i32.add
      local.get 1
      i32.store
      local.get 4
      i32.const 9
      i32.lt_u
      br_if 0 (;@1;)
      local.get 3
      local.get 1
      i32.store offset=8
      local.get 3
      local.get 1
      i32.store offset=4
      local.get 2
      i32.const -8
      i32.add
      local.get 1
      i32.store
      local.get 2
      i32.const -12
      i32.add
      local.get 1
      i32.store
      local.get 4
      i32.const 25
      i32.lt_u
      br_if 0 (;@1;)
      local.get 3
      local.get 1
      i32.store offset=24
      local.get 3
      local.get 1
      i32.store offset=20
      local.get 3
      local.get 1
      i32.store offset=16
      local.get 3
      local.get 1
      i32.store offset=12
      local.get 2
      i32.const -16
      i32.add
      local.get 1
      i32.store
      local.get 2
      i32.const -20
      i32.add
      local.get 1
      i32.store
      local.get 2
      i32.const -24
      i32.add
      local.get 1
      i32.store
      local.get 2
      i32.const -28
      i32.add
      local.get 1
      i32.store
      local.get 4
      local.get 3
      i32.const 4
      i32.and
      i32.const 24
      i32.or
      local.tee 5
      i32.sub
      local.tee 2
      i32.const 32
      i32.lt_u
      br_if 0 (;@1;)
      local.get 1
      i64.extend_i32_u
      local.tee 6
      i64.const 32
      i64.shl
      local.get 6
      i64.or
      local.set 6
      local.get 3
      local.get 5
      i32.add
      local.set 1
      loop  ;; label = @2
        local.get 1
        local.get 6
        i64.store
        local.get 1
        i32.const 24
        i32.add
        local.get 6
        i64.store
        local.get 1
        i32.const 16
        i32.add
        local.get 6
        i64.store
        local.get 1
        i32.const 8
        i32.add
        local.get 6
        i64.store
        local.get 1
        i32.const 32
        i32.add
        local.set 1
        local.get 2
        i32.const -32
        i32.add
        local.tee 2
        i32.const 31
        i32.gt_u
        br_if 0 (;@2;)
      end
    end
    local.get 0)
  (func $_start (type 3)
    (local i32)
    i32.const 65780
    i32.const 0
    memory.size
    i32.const 16
    i32.shl
    i32.const 65780
    i32.sub
    i32.const 6
    i32.shr_u
    call $memset
    drop
    local.get 0
    local.get 0
    call $command-line-arguments.main
    i32.const 0
    i32.const 1
    i32.store8 offset=65772
    call $runtime.scheduler)
  (func $command-line-arguments.main (type 5) (param i32 i32)
    i32.const 0
    call $foo
    drop
    i32.const 0
    call $foo
    drop)
  (func $runtime.scheduler (type 3)
    (local i32)
    block  ;; label = @1
      loop  ;; label = @2
        i32.const 0
        i32.load8_u offset=65772
        br_if 1 (;@1;)
        i32.const 0
        i32.load offset=65776
        local.tee 0
        i32.eqz
        br_if 1 (;@1;)
        i32.const 0
        local.get 0
        i32.load
        i32.store offset=65776
        local.get 0
        i32.const 0
        i32.store
        local.get 0
        i32.load offset=12
        local.tee 0
        local.get 0
        i32.load
        call_indirect (type 0)
        br 0 (;@2;)
      end
    end)
  (func $go_scheduler (type 3)
    call $runtime.scheduler)
  (func $runtime.lookupPanic (type 3)
    i32.const 65536
    i32.const 18
    call $runtime.runtimePanic)
  (func $runtime.runtimePanic (type 5) (param i32 i32)
    i32.const 65600
    i32.const 22
    call $runtime.printstring
    local.get 0
    local.get 1
    call $runtime.printstring
    call $runtime.printnl
    unreachable
    unreachable)
  (func $runtime.nilPanic (type 3)
    i32.const 65568
    i32.const 23
    call $runtime.runtimePanic)
  (func $runtime.printnl (type 3)
    i32.const 13
    call $runtime.putchar
    i32.const 10
    call $runtime.putchar)
  (func $runtime.putchar (type 0) (param i32)
    (local i32 i32)
    block  ;; label = @1
      i32.const 0
      i32.load offset=65632
      local.tee 1
      i32.const 119
      i32.le_u
      br_if 0 (;@1;)
      call $runtime.lookupPanic
      unreachable
    end
    i32.const 0
    local.get 1
    i32.const 1
    i32.add
    local.tee 2
    i32.store offset=65632
    local.get 1
    i32.const 65648
    i32.add
    local.get 0
    i32.store8
    block  ;; label = @1
      block  ;; label = @2
        local.get 0
        i32.const 255
        i32.and
        i32.const 10
        i32.eq
        br_if 0 (;@2;)
        local.get 1
        i32.const 119
        i32.ne
        br_if 1 (;@1;)
      end
      i32.const 0
      local.get 2
      i32.store offset=65628
      i32.const 1
      i32.const 65624
      i32.const 1
      i32.const 65768
      call $fd_write
      drop
      i32.const 0
      i32.const 0
      i32.store offset=65632
    end)
  (func $runtime.printstring (type 5) (param i32 i32)
    (local i32)
    i32.const 0
    local.set 2
    block  ;; label = @1
      loop  ;; label = @2
        local.get 2
        local.get 1
        i32.ge_s
        br_if 1 (;@1;)
        local.get 0
        local.get 2
        i32.add
        i32.load8_u
        call $runtime.putchar
        local.get 2
        i32.const 1
        i32.add
        local.set 2
        br 0 (;@2;)
      end
    end)
  (func $resume (type 3)
    call $runtime.resume$1
    unreachable)
  (func $runtime.resume$1 (type 3)
    call $runtime.nilPanic
    unreachable)
  (table (;0;) 1 1 funcref)
  (memory (;0;) 16)
  (global (;0;) (mut i32) (i32.const 65536))
  (global (;1;) i32 (i32.const 65780))
  (global (;2;) i32 (i32.const 65536))
  (global (;3;) i32 (i32.const 65780))
  (global (;4;) i32 (i32.const 65536))
  (global (;5;) i32 (i32.const 0))
  (global (;6;) i32 (i32.const 1))
  (export "memory" (memory 0))
  (export "__wasm_call_ctors" (func $__wasm_call_ctors))
  (export "memset" (func $memset))
  (export "_start" (func $_start))
  (export "__heap_base" (global 1))
  (export "command-line-arguments.main" (func $command-line-arguments.main))
  (export "go_scheduler" (func $go_scheduler))
  (export "resume" (func $resume))
  (export "__dso_handle" (global 2))
  (export "__data_end" (global 3))
  (export "__global_base" (global 4))
  (export "__memory_base" (global 5))
  (export "__table_base" (global 6))
  (data (;0;) (i32.const 65536) "index out of range\00\00\00\00\00\00\00\00\00\00\00\00\00\00nil pointer dereference\00\00\00\00\00\00\00\00\00panic: runtime error: ")
  (data (;1;) (i32.const 65624) "p\00\01\00\00\00\00\00"))
```

fgsch, 2021-02-19:
I should add, the resulting code is calling the same function twice.

deadprogram, 2021-03-03:
Now that #1661 has been merged, would it be possible to please re-test this issue against the `dev` branch? Thank you.

fgsch, 2021-03-03:
@deadprogram I will take a look shortly. 

Related question, do you know when the next release is going to be, and will it include #1661? Thanks!

deadprogram, 2021-03-03:
@fgsch has already been merged into the `dev` branch, which means it will be included in the next release, which should go out anytime in the next few days.

fgsch, 2021-03-04:
Confirmed. Revision dcff8b6 produces the expected imports.

deadprogram, 2021-03-04:
I will close this once the code is actually released. Marked with label.

fgsch, 2021-03-04:
@deadprogram ops, sorry about that!

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

PR #1661 (2021-03-03): Namespaced Wasm Imports so they don't conflict across modules, or reserved LLVM IR
closes #1643
closes #1559

Kind of relates to #1647
Doesn't fix, but puts on the right path I think, for relates to #1648

So this PR was opened from a discussion in the tinygo slack: https://gophers.slack.com/archives/CDJD3SUP6/p1614201204223100 . Involving a conflict with wasm imports with the name of `write`. Which conflicts with the LLVM IR `@write`, and requires decoupling the LLVM IR symbol, and the `wasm-import-name`.

I tested this pr with the following code: https://github.com/torch2424/tinygo-compiler- [... cut]
closed 2021-03-08 by deadprogram