tinygo-org/tinygo #1648: Cannot define fd_write in an imported package

fgsch, 2021-02-21:
Due to constraints in my environment, `fd_write` needs to be imported from `wasi_snapshot_preview1`.
This is possible if it is defined in the main package, for example using the following:

```go
//go:wasm-module wasi_snapshot_preview1
//export fd_write
func placeholder() int32
```

or, using a similar signature:

```go
//go:wasm-module wasi_snapshot_preview1
//export fd_write
func fd_write(int32, *int32, int32, *int32) int32
```

But using either of these approaches in an imported package fails with:

```
Attribute after last parameter!
i32 ()* @fd_write
Incorrect number of arguments passed to called function!
  %3 = call i32 @fd_write(i32 1, %runtime.__wasi_iovec_t* @runtime.putcharIOVec, i32 1, i32* @runtime.putcharNWritten)
error: verification error after IR construction
```

or:

```
Call parameter type does not match function signature!
%runtime.__wasi_iovec_t* @runtime.putcharIOVec
 i32*  %3 = call i32 @fd_write(i32 1, %runtime.__wasi_iovec_t* @runtime.putcharIOVec, i32 1, i32* @runtime.putcharNWritten)
error: verification error after IR construction
```

respectively. 
I'm not sure if there is a way to match the signature as `runtime.__wasi_iovec_t` is not exported.

fgsch, 2021-02-21:
Related to #1647.

aykevl, 2021-03-05:
It seems to me that the correct solution is #1647, which should be straightforward to fix.

fgsch, 2021-03-15:
@aykevl Agreed. Let's close this one.

closed 2021-03-28 by deadprogram