tinygo-org/tinygo #1651: WASM: panic: trying to make exported function async

Nerzal, 2021-02-22:
I get an error when trying to build a wasm project with an exported function.
I tried the ` // export functionName` syntax as described in the [documentation](https://tinygo.org/webassembly/webassembly/)

Is there a way to prevent the compiler toolchain from trying to make these optimizations? 

TinyGo version: 
`tinygo version 0.17.0-dev linux/amd64 (using go version go1.16 and LLVM version 11.1.0)`

Error:
```bash
panic: trying to make exported function async: messageHandler

goroutine 11 [running]:
github.com/tinygo-org/tinygo/transform.(*coroutineLoweringPass).findAsyncFuncs(0xc0034079e0)
        /home/tobias/go/src/github.com/Nerzal/tinygo/transform/coroutines.go:186 +0x997
github.com/tinygo-org/tinygo/transform.(*coroutineLoweringPass).load(0xc0034079e0, 0x20, 0x7fe51e2cbb30)
        /home/tobias/go/src/github.com/Nerzal/tinygo/transform/coroutines.go:305 +0x2ff
github.com/tinygo-org/tinygo/transform.LowerCoroutines(0x7fe50c002b30, 0x7fe50c002b01, 0x0, 0x0)
        /home/tobias/go/src/github.com/Nerzal/tinygo/transform/coroutines.go:70 +0x139
github.com/tinygo-org/tinygo/transform.Optimize(0x7fe50c002b30, 0xc00021a510, 0x2, 0x2, 0x5, 0x0, 0x0, 0x0)
        /home/tobias/go/src/github.com/Nerzal/tinygo/transform/optimizer.go:120 +0xee5
github.com/tinygo-org/tinygo/builder.compileWholeProgram(0x7fff770e1bda, 0x17, 0xc00021a510, 0xc0001526e0, 0xc000152790, 0x23e2750, 0xe9f5ed9, 0xe9f5ed900000000, 0x603382ac)
        /home/tobias/go/src/github.com/Nerzal/tinygo/builder/build.go:419 +0x61c
github.com/tinygo-org/tinygo/builder.Build.func1(0xc004fe8b0e9f5ed9, 0xfb1421d)
        /home/tobias/go/src/github.com/Nerzal/tinygo/builder/build.go:97 +0x72
github.com/tinygo-org/tinygo/builder.jobWorker(0xc000102600, 0xc0001025a0)
        /home/tobias/go/src/github.com/Nerzal/tinygo/builder/jobs.go:153 +0xdf
created by github.com/tinygo-org/tinygo/builder.runJobs
        /home/tobias/go/src/github.com/Nerzal/tinygo/builder/jobs.go:73 +0xc5
```

Exported function: 

```go
//export messageHandler
func messageHandler(message string) {
	println("mqtt: message received")

	splittedStrings := strings.Split(message, "#")

	temperature, err := strconv.ParseFloat(splittedStrings[4], 32)
	if err != nil {
		println("failed to parse temperature from message")
		println(message)
	}

	pressure, err := strconv.ParseFloat(splittedStrings[6], 32)
	if err != nil {
		println("failed to parse pressure from message")
		println(message)
	}

	humidity, err := strconv.ParseFloat(splittedStrings[8], 32)
	if err != nil {
		println("failed to parse humidity from message")
		println(message)
	}

	mqttMessages <- SensorEvent{
		TimeStamp:   splittedStrings[0],
		Message:     splittedStrings[2],
		Temperature: float32(temperature),
		Pressure:    float32(pressure),
		Humidity:    float32(humidity),
	}
```
}

Nerzal, 2021-02-22:
Using ` -opt 1` build flag resolves the issue. 
But we might still want to fix that behavior using the default flags.

Nerzal, 2021-02-22:
So the problem seems to be, that this function wants to write something into a channel

aykevl, 2021-03-05:
> Is there a way to prevent the compiler toolchain from trying to make these optimizations?

No. Some operations (such as sending to a channel) block. At the same time, an exported function called by JavaScript cannot block. Therefore it is not possible to do any blocking operations in an exported function.

That said, the error message is not very helpful and could be improved a lot to help users debug this issue (which is quite common).

aykevl, 2021-03-05:
I have created a PR that will make the error a lot better: https://github.com/tinygo-org/tinygo/pull/1688

Nerzal, 2021-03-05:
> Some operations (such as sending to a channel) block. At the same time, an exported function called by JavaScript cannot block. Therefore it is not possible to do any blocking operations in an exported function.

We might want to add that to the documentation :+1: 

Thanks for explaining and improving the error message!

Nerzal, 2021-03-05:
If issues get automatically closed, when they have the "nextrelease" tag, this would be a candidate for that :)

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram