tinygo-org/tinygo #1652: WASM:  syscall/js.finalizeRef not implemented

Nerzal, 2021-02-22:
I do actually see 2 errors, and i'm not quite sure, if the first one is the cause of the panic. 

The first error is a syscall/js.finalizeRef not implemented. I'm not even sure, which part of the JS could even call a finalizer, as i do not have a really good understanding of javascript. 

[image]

The second error is then an index out of range
[image]

TinyGo Version: 

> tinygo version 0.17.0-dev linux/amd64 (using go version go1.16 and LLVM version 11.1.0)

Exported function by:

```javascript
	js.Global().Set("messageHandler", js.FuncOf(MessageHandler))
```

Call in js:
```javascript
    messageHandler(message.payloadString)
```

String being passed as payloadString: 
> 2021-02-21T23:39:59+01:00#Sensor reading#24.32#1000.15#54.32

Function in wasm.go

```go
func MessageHandler(this js.Value, args []js.Value) interface{} {
	println("mqtt: message received")

	message := args[0].String()

	splittedStrings := strings.Split(message, "#")

	temperature, err := strconv.ParseFloat(splittedStrings[4], 32)
	if err != nil {
		println("failed to parse temperature from message")
		println(message)
	}

	pressure, err := strconv.ParseFloat(splittedStrings[6], 32)
	if err != nil {
		println("failed to parse pressure from message")
		println(message)
	}

	humidity, err := strconv.ParseFloat(splittedStrings[8], 32)
	if err != nil {
		println("failed to parse humidity from message")
		println(message)
	}

	mqttMessages <- SensorEvent{
		TimeStamp:   splittedStrings[0],
		Message:     splittedStrings[2],
		Temperature: float32(temperature),
		Pressure:    float32(pressure),
		Humidity:    float32(humidity),
	}

	return nil
}
```

deadprogram, 2021-02-24:
See discussion at https://github.com/tinygo-org/tinygo/issues/1140

Seems like still needs to be done! :smile_cat:

Nerzal, 2021-02-24:
Oh sorry, should have searched first

deadprogram, 2021-02-24:
Not at all, that info was a little buried unless doing a text search. I only did, because I recalled it existed. :mag_right: