tinygo-org/tinygo #1653: LLVM ERROR: Cannot select: intrinsic %llvm.coro.size

ALTree, 2021-02-22:
```
$ tinygo version
tinygo version 0.16.0 linux/amd64 (using go version go1.15.8 and LLVM version 10.0.1)
```
```
package main

func main() {
	var ch chan string
	var m map[byte]string
	_ = m[0]
	_ = <-ch
}
```
```
$ tinygo build -o out crash.go 
LLVM ERROR: Cannot select: intrinsic %llvm.coro.size
```

deadprogram, 2021-02-24:
Is that actually valid Go code?

```
fatal error: all goroutines are asleep - deadlock!

goroutine 1 [chan receive (nil chan)]:
main.main()
	/tmp/sandbox401551102/prog.go:7 +0x65

Program exited: status 2.
```

aykevl, 2021-02-24:
It is valid in the sense that it is well defined: a program like this is defined to exit with a panic. This is certainly a compiler bug, either in TinyGo or in LLVM.

ALTree, 2021-02-24:
@deadprogram Yes, it's valid, in the sense that it conforms to the [Go Specification](https://golang.org/ref/spec). 

As @aykevl pointed out, in general a compiler shouldn't crash when compiling valid -or invalid- Go code. Valid code should generate a binary, which then may or may not panic at runtime, like mine does; invalid Go code should print a compilation error. In this case LLVM is crashing while processing a valid Go program, so this can be classified as a compiler bug.

niaow, 2021-12-14:
This should be addressed now that coroutines are no longer used (and will be removed by #2174 )

ALTree, 2022-08-20:
Confirmed fixed on 0.25

closed 2022-08-20 by ALTree