tinygo-org/tinygo #1658: Tinygo on Linux uses go version 0.1

allthingssecurity, 2021-02-24:
When I try to run a tinygo program it says it is using version 0.1 of go instead of something between 1.11 and 1.15.
I tried deb package of both tinygo 0.15 and 0.16 but I get same error

deadprogram, 2021-02-24:
How exactly did you produce that version? I ask because normally you can tell by running the command `tinygo version` like this:

```shell
$ tinygo version
tinygo version 0.17.0-dev-418ab8b linux/amd64 (using go version go1.15 and LLVM version 11.0.0)
```

I am not aware of any other way to do so.

Raqbit, 2021-02-24:
When you set the tinygo cached GOROOT (as found by `tinygo info <board>`) as your GOROOT in GoLand, it also shows `v0.1.0` as the go version, not sure why, but probably related.

aykevl, 2021-02-24:
In that case I think GoLand picks up the wrong version number.

TinyGo will use the Go installation that it finds by resolving the `go` command. I can assure you that it won't magically use Go 0.1, which is now very old.

Raqbit, 2021-02-26:
@aykevl Found it, Goland picks up the version number from `src/runtime/internal/sys/zversion.go` in `GOROOT` via a regex.

Found it [here](https://github.com/go-lang-plugin-org/go-lang-idea-plugin/blob/e73dab44213ffa76b3d6a853aecb109929c3e2b5/src/com/goide/sdk/GoSdkUtil.java#L372) in the legacy open source go-lang-idea-plugin, the code of which probably still lives in GoLand.

aykevl, 2021-03-11:
Apparently I did not really understand the issue.

> When you set the tinygo cached GOROOT (as found by `tinygo info <board>`) as your GOROOT in GoLand

It looks like GoLand then also passes this cached GOROOT on to TinyGo, which expects the root of the Go installation instead of this cached GOROOT. See my comment on #1705 for more details.
This is indeed a bug. I could say that GoLand shouldn't pass on this GOROOT but that's actually sensible behavior. Instead, I think TinyGo should recognize this situation and deal with it (again, as detailed in #1705).

deadprogram, 2021-09-18:
This appears to be have been resolved. Please reopen if needed.

closed 2021-09-18 by deadprogram