tinygo-org/tinygo #1662: How to do tinygo-gdb on windows

sago35, 2021-02-26:
The following changes make me use gdb-multiarch, but I still think I need to use arm-none-eabi-gdb.exe on windows.

* 1a2f051b

If rename arm-none-eabi-gdb.exe to gdb-multiarch.exe, it can be debugged with tinygo-gdb.
However, I think it is better to specify it with an argument like `tinygo gdb -debugger=arm-none-eabi-gdb ...` rather than renaming it.
Since I already created something that works, I will create a Pull-Request if this approach is okay.

or,,,
Is there any other better way?

windows 10 + tinygo 0.16 or later + cortex-m4

```
error: failed to run gdb with C:\Users\sago3\AppData\Local\Temp\tinygo160016967\main: exec: "gdb-multiarch": executable file not found in
 %PATH%
```

arm-neno-eabi-gdb for windows can be found at:
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

aykevl, 2021-02-27:
>  However, I think it is better to specify it with an argument like `tinygo gdb -debugger=arm-none-eabi-gdb ...` rather than renaming it.

What do you think of checking both? That avoids a flag the user would have to pass to debug something. For example, by making this a slice instead of a string and checking all commands:

https://github.com/tinygo-org/tinygo/blob/66f76833ad0ac1dd31befa3ecbd8c0d1e5bb5ca3/compileopts/target.go#L45

(There is [`exec.LookPath`](https://pkg.go.dev/os/exec#LookPath) to check whether a command exists).

sago35, 2021-02-28:
I think it's a very good idea.
I've created a pull request ( #1670 )

sago35, 2021-02-28:
@aykevl  cc: @deadprogram 
I' m also planning to create the following pull requests.
I hope to have them ready for TinyGo 0.17 if possible.

* Getting tinygo gdb to work on windows
    * #1670 : gdb: enable to specify multiple candidates for gdb 
    * Support daemonization
        * Now when I press Ctrl-C, openocd exits.
    * fix `tinygo dbg (openocd) does not exit` #1366 
    * -ocd-output will be colored on windows.

sago35, 2021-03-05:
Once the aforementioned pull-request is merged, tinygo-gdb will work normally on Windows.

sago35, 2021-03-05:
I forgot to mention that I also need the following.

* #1397 : flash: add verify on openocd-flash

deadprogram, 2021-09-07:
Can this issue now be closed @sago35 ?

sago35, 2021-09-07:
Yes, we can run tinygo gdb on windows now.
Since #1397 is not related to this issue, it can be considered separately.

PR #1670 (2021-03-28): gdb: enable to specify multiple candidates for gdb
fixes #1662 (How to do tinygo-gdb on windows)
closed 2021-09-07 by sago35