tinygo-org/tinygo #1664: Why this project stopped?

vidalouiswang, 2021-02-27:


Nerzal, 2021-02-27:
Could you please elaborate what exactly you intend to ask?

deadprogram, 2021-02-27:
@vidalouiswang perhaps you are looking for the active `dev` branch:

https://github.com/tinygo-org/tinygo/tree/dev

vidalouiswang, 2021-02-27:
> Could you please elaborate what exactly you intend to ask?

> Could you please elaborate what exactly you intend to ask?

Hi,
I just noticed this project haven't any commit for long time, so want to know some information.

vidalouiswang, 2021-02-27:
> @vidalouiswang perhaps you are looking for the active `dev` branch:
> 
> https://github.com/tinygo-org/tinygo/tree/dev

Thank you, I am a beginner, I searched on the offical website and saw tinygo doesn't support wifi on ESP8266 and ESP32.
I'm interest in tinygo project, and I would like to help to create new features. But I dont't think I can create any code now.
So I want to know is there any clue to start with? Like where should I learn from? Provide some links if possible, thank you.

closed 2021-02-27 by vidalouiswang