tinygo-org/tinygo #1666: Harmonize STM32 tick / sleep logic

kenbell, 2021-02-27:
As part of discussion on #1608 it looks like the STM32 tick / sleep logic is diverged and has bugs.  This issue is to track the proposal to harmonize on a single implementation of tick/sleep for STM32 MCUs.

---

Taking a quick survey, there are two patterns:

### Algorithm 1

1. Set timer interrupt for at most 'd' (timer may not have resolution to wait for whole of 'd')
2. Wait for the timer interrupt (ignoring other exits from wfi)
3. Return to scheduler

```
runtime_stm32f7x2.go
runtime_stm32f405.go
runtime_stm32f407.go
runtime_stm32l5x2.go
runtime_stm32f103.go (my change)
```

### Algorithm 2

1. Set timer interrupt for at most 'd' (timer may not have resolution to wait for whole of 'd')
2. Wait for the timer interrupt (ignoring other exits from wfi)
3. If timer didn't have resolution for full sleep duration, go back to step 1
4. Return to scheduler

```
runtime_stm32l0.go
runtime_stm32f103.go (original, before #1466)
```

### Proposal

1. Revert change #1466 to unbreak sleep in the short-term (i.e. what @aykevl proposes)
2. Specify the expected semantics of sleepTicks precisely
3. Harmonize the STM32 sleep functionality to prevent divergence (with the exception of stm32f103 they all use a timer, only stm32f103 uses RTC - they call all use timers?)

My understanding of the spec for sleepTicks is:
```
If hasScheduler
  Sleep for at least the specified duration (allowing for overhead of setting up the sleep) unless any interrupt fires (including the timer interrupt), in which case return to the scheduler

Else (ie hasScheduler = false)
  Sleep for at least the specified duration (allowing for overhead of setting up the sleep), do not return early regardless of any interrupts that fire
```

### Bugs

If I've understood correctly, there's an additional bug in #1466 that if two interrupts had fired (the timer interrupt + one other), the logic assumed only the timer interrupt had fired and would not return to the scheduler early.

Less significant, the logic in `runtime32l0.go` also accumulates errors if it goes back to timerSleep multiple times since it doesn't recalculate from a fixed reference time - it just subtracts the 'asked for sleep duration' from the 'total sleep duration' not allowing for overheads.

### Harmonizing STM32 sleep

Given the subtlety around the sleep logic, it really seems like we should have a single implementation for STM32 chips - the basic timers are essentially identical on the chips i did a quick look at, so it should be possible to have a common implementation.  Part of this would be to move the chips that use the RTC for 'ticks' to use a timer.

It looks like it should be possible to eliminate 80% of the code from the `runtime_stm32xxxx.go` files.

aykevl, 2021-02-27:
Overall, I agree with your proposal. Just a few remarks here and there.

> Given the subtlety around the sleep logic, it really seems like we should have a single implementation for STM32 chips - the basic timers are essentially identical on the chips i did a quick look at, so it should be possible to have a common implementation.

Agreed. The more code that can be shared, the better (unless code sharing introduces additional complexity).

>  Part of this would be to move the chips that use the RTC for 'ticks' to use a timer.

I have not checked the stm32, but usually an RTC uses less current than a timer. I know this is true on Nordic chips: the timers use many times more energy than the RTC. That's why I have used the RTC on Nordic chips.
That said, chip families and especially chip vendors often have lots of difference and even use the same word for different kinds of peripherals. Therefore it would be a good idea to check how this works on stm32 chips.

> Less significant, the logic in `runtime32l0.go` also accumulates errors if it goes back to timerSleep multiple times since it doesn't recalculate from a fixed reference time - it just subtracts the 'asked for sleep duration' from the 'total sleep duration' not allowing for overheads.

Usually the timers are not nearly precise enough that this matters. Also, it only happens over longer periods of time (otherwise the timer wouldn't overflow). I do not think this is a real problem, but if it's possible to recalculate from the current time that would be better.
However, keep in mind that it is okay to sleep a little bit longer than intended, but is never okay to sleep for a shorter period of time. Sleeping for a shorter period than specified in `time.Sleep` is a bug. In particular, watch out for integer rounding which rounds down.

And last, I've seen that some implementations use `wfi` for "wait for interrupt". Unfortunately that introduces a race condition (the interrupt might have fired already right before the `wfi` instruction), it's better to use `wfe` instead.

kenbell, 2021-02-28:
I've put up a draft PR to harmonize the logic.  On the whole I think it's pretty clean.

I decided against using RTC because that seems most valuable reserved for deep sleep states, using the RTC to hold wall time and/or using RTC alarms to awake from deep sleep.  Current use of RTC for 'ticks' prevents use in that way.

I need some advice about STM32L0.  The ARR field is 16-bit in `TIM_type` on that architecture.  Most all timers are 16-bit on all architectures, but `dev/stm32` seems to expand to `Register32` because most architectures have 32-bit timers as well.  However, STM32L0 doesn't appear to have a 32-bit timer, so the expansion doesn't occur.

Not sure the best way to resolve:

- Have two copies of the harmonized logic, one for 16-bit ARR, one for 32-bit ARR
- Fix `gen-device-stm32` so that 16-bit and 32-bit timers aren't merged into a single Go type
- Have a per chip function `convertToARRtype` which converts 32-bit values to whichever type ARR register expects

aykevl, 2021-03-04:
> I decided against using RTC because that seems most valuable reserved for deep sleep states, using the RTC to hold wall time and/or using RTC alarms to awake from deep sleep.

Isn't that exactly what `time.Sleep` should do? Set the chip to a low power state until it is ready to wake up.

> I need some advice about STM32L0. The ARR field is 16-bit in `TIM_type` on that architecture. Most all timers are 16-bit on all architectures, but `dev/stm32` seems to expand to `Register32` because [...]

The SVD file lists a 32-bit ARR register with a 16-bit ARR field. As far as I can see, gen-device-svd does exactly what it is supposed to do.

kenbell, 2021-03-06:
At the very lowest power levels, RAM is powered down and can be limited to just a few wake-up sources, one of which is RTC:
[image]

To support timed wake-up from these power states, I think it's best to reserve RTC for that.  The semantics wouldn't match sleep since a bunch of re-initialization is required (not least of memory state).  The current code resets RTC to zero on initialization, which excludes use-cases where you want to use a battery / capacitor to maintain 'wall clock' across resets and timed wake-ups (maybe a sensor use-case using RTC to wake-up every few minutes to take a sample?).  I'm assuming implementation-wise this kind of power state would be handled in TinyGo by treating it as-if there was a processor reset, so would initialize the runtime and call 'main'.

For cases where RAM state is preserved and the processor can be woken from a number of external peripherals, I think it would be possible to build that functionality into the semantics of 'sleep' - but I'm treating as outside of the scope of this change for now :)

---

On the ARR field, I've taken a further look.  For the stm32l0x2, the TIM6 ARR field is this in the SVD:
```
        <register>
          <name>ARR</name>
          <displayName>ARR</displayName>
          <description>auto-reload register</description>
          <addressOffset>0x2C</addressOffset>
          <size>0x20</size>
          <access>read-write</access>
          <resetValue>0x00000000</resetValue>
          <fields>
            <field>
              <name>ARR</name>
              <description>Low Auto-reload value</description>
              <bitOffset>0</bitOffset>
              <bitWidth>16</bitWidth>
            <writeConstraint><range><minimum>0</minimum><maximum>65535</maximum></range></writeConstraint>
            </field>
          </fields>
        </register>
```

32-bit register, with 16-bit field - but in the stm32l0x2.go TIM6 maps to TIM_Type, which is:
```
// General-purpose-timers
type TIM_Type struct {
	CR1          volatile.Register32 // 0x0
	CR2          volatile.Register32 // 0x4
	SMCR         volatile.Register32 // 0x8
	DIER         volatile.Register32 // 0xC
	SR           volatile.Register32 // 0x10
	EGR          volatile.Register32 // 0x14
	CCMR1_Output volatile.Register32 // 0x18
	CCMR2_Output volatile.Register32 // 0x1C
	CCER         volatile.Register32 // 0x20
	CNT          volatile.Register16 // 0x24
	_            [2]byte
	PSC          volatile.Register32 // 0x28
	ARR          volatile.Register16 // 0x2C
	_            [6]byte
	CCR1         volatile.Register16 // 0x34
	_            [2]byte
	CCR2         volatile.Register16 // 0x38
	_            [2]byte
	CCR3         volatile.Register16 // 0x3C
	_            [2]byte
	CCR4         volatile.Register16 // 0x40
	_            [6]byte
	DCR          volatile.Register32 // 0x48
	DMAR         volatile.Register32 // 0x4C
	OR           volatile.Register32 // 0x50
}
```

For the STM32F406, for TIM6 the SVD is this:
```
        <register>
          <name>ARR</name>
          <displayName>ARR</displayName>
          <description>auto-reload register</description>
          <addressOffset>0x2C</addressOffset>
          <size>0x20</size>
          <access>read-write</access>
          <resetValue>0x00000000</resetValue>
          <fields>
            <field>
              <name>ARR</name>
              <description>Low Auto-reload value</description>
              <bitOffset>0</bitOffset>
              <bitWidth>16</bitWidth>
            <writeConstraint><range><minimum>0</minimum><maximum>65535</maximum></range></writeConstraint>
            </field>
          </fields>
        </register>
```
It maps to TIM_Type, which has a 32-bit ARR:
```
// Advanced-timers
type TIM_Type struct {
	CR1          volatile.Register32 // 0x0
	CR2          volatile.Register32 // 0x4
	SMCR         volatile.Register32 // 0x8
	DIER         volatile.Register32 // 0xC
	SR           volatile.Register32 // 0x10
	EGR          volatile.Register32 // 0x14
	CCMR1_Output volatile.Register32 // 0x18
	CCMR2_Output volatile.Register32 // 0x1C
	CCER         volatile.Register32 // 0x20
	CNT          volatile.Register32 // 0x24
	PSC          volatile.Register32 // 0x28
	ARR          volatile.Register32 // 0x2C
	RCR          volatile.Register32 // 0x30
	CCR1         volatile.Register32 // 0x34
	CCR2         volatile.Register32 // 0x38
	CCR3         volatile.Register32 // 0x3C
	CCR4         volatile.Register32 // 0x40
	BDTR         volatile.Register32 // 0x44
	DCR          volatile.Register32 // 0x48
	DMAR         volatile.Register32 // 0x4C
}
```

I may be reading the SVD incorrectly, but it looks like essentially similar register definitions are resulting in different Go structures.

kenbell, 2021-03-13:
For now, I've resolved the ARR type differences by having a type alias that is either uint32 or uint16, defined per MCU.

If possible, i'd like to get this merged fairly soon so it has a long time to 'soak' in dev since it changes a critical part of all the code for all STM32 MCUs

kenbell, 2021-03-13:
I've hooked up a multi-meter to a bluepill (stm32f103).  Using blinky1 example, both the dev branch (using RTC) and this branch (using timer) pull the same power as far as I can tell ranging 15mA - 17mA, so I'm not seeing a significant difference in power draw. (at least that my kit is able to measure)

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

closed 2021-05-12 by deadprogram