tinygo-org/tinygo #1671: WASM big int support

hamidrezakks, 2021-02-28:
Hi,

Well I've asked this in a comment, now I decided to issue it separately.

I get this error while using bigInt for cryptography purposes
```
WebAssembly.instantiate(): Import #5 module="env" function="math/big.addVV" error: function import requires a callable
is math/bigInt supported.
```

Cloud you please check this out,
It seems WASI is using native JS big int,
Is it possible to resolve this within WASM built-ins?

Best KK

aykevl, 2021-03-04:
Can you try adding `-tags=math_big_pure_go` to the command? I think it will fix this issue.

hamidrezakks, 2021-03-05:
Thanks, it fixed the big int issue, but I faced a new issue:
```syscall/js.finalizeRef not implemented```
I guess I need to wait for this feature

fgsch, 2021-03-15:
I think this is a dup of #1140 at this point.

closed 2021-05-22 by hamidrezakks