tinygo-org/tinygo #1672: Drop support for Go 1.11 and Go 1.12

deadprogram, 2021-02-28:
In the past we have had reasons for maintaining support for older versions. I think this was mostly because of the Go version in the package manager for LTS versions of Debian.

Now I wonder if we should just maintain the exact same support schedule as Go itself, plus one extra to accommodate that we are usually slightly lagging the release of new Go major versions.

If we do that, then it would mean that we could be dripping support for 1.11 and 1.12 right away.

As part of this we would also drop support for LLVM 9.

One benefit besides the extra effort to maintain compatibility, is that we would have 2 less CI builds to require processing, speeding up the build process.

What do we think of this?

Nerzal, 2021-02-28:
I don't see any reason, why we should not just do it. 
Even if package managers of any LTS versions of some derivates lag behind, Go can still be manually downloaded and installed, which is the way i update my Go version.

aykevl, 2021-02-28:
Yes, agreed. Perhaps we can do this after the coming release?

deadprogram, 2021-02-28:
We can do it after this next release. That probably means we will not be able to merge https://github.com/tinygo-org/tinygo/pull/1659 until the next release.

ofauchon, 2021-03-01:
@Nerzal is right. Go is really to install locally and upgrade.
We don't see any reasons to continue supporting Go 1.11 and Go 1.12.
Move forward and go 1.13 !

deadprogram, 2021-03-02:
I have started the process here https://github.com/tinygo-org/tinygo/commit/c450ec940358c5424417ace7751c7ca8d8d1d8a1

deadprogram, 2021-03-02:
Not sure why the Go 1.11/LLVM 9 build has been failing: https://app.circleci.com/pipelines/github/tinygo-org/tinygo/4439/workflows/e778be24-c568-4776-a96b-2c6b4aaa94f4/jobs/24630

Perhaps we need to accelerate our plans here?

deadprogram, 2021-03-04:
CircleCI build still failing, seeming because of something that is no longer available for download from LLVM for the specific set of other dependencies for LLVM v9.

via the LLVM site:

> Packages are available for amd64 and i386 (except for recent Ubuntu) and for both the stable, old-stable and development branches (currently 11, 12 and 13).

All the more reason to do this. I am starting to lean towards just doing it now.

deadprogram, 2021-03-04:
Looking at the CircleCI error log a little more closely:

```
N: Repository 'http://deb.debian.org/debian buster InRelease' changed its 'Version' value from '10.0' to '10.8'
```

I think the issue is that CircleCI needs to update their pre-built Docker image that we are using as a base, and that it is so old that they have not done so as part of their normal update process.

trevor403, 2021-03-09:
Is the above PR ready to merge now that 0.17.0 has released?

aykevl, 2021-03-09:
In my opinion, yes.

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

closed 2021-05-12 by deadprogram