tinygo-org/tinygo #1676: Open aways fails on macOS

EX0l0N, 2021-03-02:
I was trying to shrink a small binary I wrote using tinygo. Not sure if darwin / macOS is even a supported target for tinygo, I just did the same thing I would do on linux (not specifying any target architecture) and compiled.

I got a binary, but running it instantly failed with `Erno 78` while trying to read from `stdin`.

Trying to isolate the problem I wrote a small main func that only opens a file for reading. Again, Errno 78.

Even using the already connected `fd 0` via syscall.Read has the same result.

If this is to be expected because macOS is not a supported target (it compiles/runs hello world though), please close.

Otherwise, I think something as basic as `open` should be supported.

deadprogram, 2021-03-03:
Hello @EX0l0N please take a look at https://github.com/tinygo-org/tinygo/pull/1659 which should start to add some of the macOS support you are looking for.

EX0l0N, 2021-03-04:
I see… .

So this problem is really existing, know and there are people working on it.

Not need for this issue to be open then. thx @deadprogram .

closed 2021-03-04 by EX0l0N

aykevl, 2021-03-04:
We do have some support for macOS, but definitely not full support. Especially I/O is something that's not yet well developed in TinyGo.

EX0l0N, 2021-03-04:
That's fine I was just curious if there is any support at all. Neither the term macOS nor darwin show up somewhere in the official target list.

I was already surprised, that I could compile a simple hello world without specifying any cross target.

From time to time I will give it a try again and see where you're at.