tinygo-org/tinygo #1678: Document make wasi-libc

fgsch, 2021-03-04:
This is not mentioned anywhere I can see. It would be a good idea to document it.

aykevl, 2021-03-04:
Please take a look at https://github.com/tinygo-org/tinygo/pull/1679 which should make the error message a lot clearer and more helpful. For example, I got this error message after removing the libc:

```
$ go install; and tinygo run -target=wasi ./testdata/cgo
error: could not find wasi-libc, perhaps you need to run `make wasi-libc`?
```

That's a lot more helpful than a linker error.

fgsch, 2021-03-04:
@aykevl I will take a look. Thank you.

Can we still document this step in the building instructions?

aykevl, 2021-03-04:
Done: https://github.com/tinygo-org/tinygo-site/pull/150

fgsch, 2021-03-04:
Re #1679, works as advertised. Thank you!

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram