tinygo-org/tinygo #1686: map[string]uint64 generates incorrect wasm?

dkegel-fastly, 2021-03-05:
Using
```
tinygo version 0.16.0 darwin/amd64 (using go version go1.15.3 and LLVM version 10.0.1)
```

The innocent little program
```
package main
var foo32 = map[string]uint32{
	"foo": 7,
}
var foo64 = map[string]uint64{
	"foo": 7,
}
func main() {
	if foo32["foo"] != 7 {
		panic("unexpected")
	}
	if foo64["foo"] != 7 {
		panic("map cannot hold uint64 in wasm?")
	}
}
```
works fine with non-tiny go or default tinygo, but fails if built and run like so:
```
tinygo build -target=wasi -scheduler=none -wasm-abi=generic -o badness .
wasmtime badness
```
I think it fails with another wasm runtime, too, so it kind of feels like a compiler bug.

aykevl, 2021-03-05:
That's indeed a bad compiler bug. However, it seems to have been fixed with https://github.com/tinygo-org/tinygo/pull/1430 (tested with bisect) and I can no longer reproduce it on the dev branch. Can you try the dev branch?

dkegel-fastly, 2021-03-05:
I tried to follow the instructions for building on mac os 10.15.7 with ```brew install llvm && go install```, but ran into https://github.com/tinygo-org/tinygo/issues/1633

aykevl, 2021-03-05:
We'll have a release today.

dkegel-fastly, 2021-03-05:
Well, poot.  After building tinygo with 'go install', I got:
```
$ ~/go/bin/tinygo build -target=wasi -scheduler=none -wasm-abi=generic -o badness .
error: could not find wasi-libc, perhaps you need to run `make wasi-libc`?
```
guess maybe you don't get to use brew's llvm if you want wasm support?

aykevl, 2021-03-05:
Have you tried running the command as suggested in the error message? WebAssembly should definitely work when using LLVM from Homebrew.

aykevl, 2021-03-05:
(you may need to run `git submodule update --init` before it will work)

dkegel-fastly, 2021-03-06:
When I tried, "make wasi-libc" failed with 
```error: unable to create target: 'No available targets are compatible with triple "wasm32-unknown-wasi"'```
which meant I didn't have /usr/local/opt/llvm/bin first in my PATH.
After updating PATH, make succeeded.

However, the resulting tinygo compiler got a SIGSEGV:
```
tinygo.org/x/go-llvm._Cfunc_LLVMGetLastParam(0x3a5047a8, 0x0)
	_cgo_gotypes.go:5892 +0x4a fp=0xc00140f868 sp=0xc00140f838 pc=0x424a70a
tinygo.org/x/go-llvm.Value.LastParam.func1(0x3a5047a8, 0x3a5050a0)
	/Users/dkegel/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210206225315-7fe719483a0f/ir.go:1213 +0x4d fp=0xc00140f898 sp=0xc00140f868 pc=0x42647ed
```
Looks like go 1.15.3, https://github.com/golang/go/issues/42032.  
I manually removed the .o's and .a's related to wasi-libc, repeated the make, and repeated the go install.

Now tinygo insists on a go.mod, so I added a trivial one.

Whew.  After all that, yes, I can confirm that the new build seems to fix the problem.

But holy papercut, batman, that took some fiddling :-)

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram

aykevl, 2021-03-08:
I'm sorry you had to go through all of this :( It looks like you somehow managed to hit most of the MacOS issues you could have been hitting.

dkegel-sigsci, 2021-03-09:
I knew the job was dangerous when I took it!