tinygo-org/tinygo #1697: Potential Bug Risks and Anti-Patterns

HarshCasper, 2021-03-08:
## Description 

Hi 👋 

I ran DeepSource Static Code Analysis upon the Project, the results for which are available [here](https://deepsource.io/gh/HarshCasper/tinygo/). 

The Static Code Analysis Tool found potential bugs and anti-patterns in the Code, that can be detrimental at a later point in time with respect to the Project. DeepSource helps you to automatically find and fix issues in your code during code reviews. This tool looks for anti-patterns, bug risks, performance problems, and raises issues. 

Some of the notable issues are: 

- Unused Code [here](https://deepsource.io/gh/HarshCasper/tinygo/issue/SCC-U1000/)
- Slice can be used directly [here](https://deepsource.io/gh/HarshCasper/tinygo/issue/CRT-A0016/)
- Empty String Tests [here](https://deepsource.io/gh/HarshCasper/tinygo/issue/CRT-A0004/)
- Potential trucation issue fond [here](https://deepsource.io/gh/HarshCasper/tinygo/issue/CRT-D0020/)
- Empty fall through [here](https://deepsource.io/gh/HarshCasper/tinygo/issue/CRT-A0003/)

There are plenty of other issues in relation to Bug Discovery and Anti-Patterns which you would be interested to take a look at.

If you would like to integrate DeepSource to autofix some of the commonly occurring issues, I can help you set that up :)

aykevl, 2021-03-08:
> * Unused Code [here](https://deepsource.io/gh/HarshCasper/tinygo/issue/SCC-U1000/)

All of this code is in fact used, but the analyzer is unable to see this. Therefore these are false positives.

> * Slice can be used directly [here](https://deepsource.io/gh/HarshCasper/tinygo/issue/CRT-A0016/)

There are two true positives, the other 124 issues are false positives: cgo/testdata contains test and test output files for the compiler and so should not be included in the analysis.

> * Empty String Tests [here](https://deepsource.io/gh/HarshCasper/tinygo/issue/CRT-A0004/)

That's an interesting one. The two are semantically equivalent but are not currently optimized the same way in TinyGo compiled code. Therefore I've created a PR to fix that: https://github.com/tinygo-org/tinygo/pull/1698

I haven't yet looked into other flagged issues.

dgryski, 2023-01-31:
I looked at all the `fallthrough`s -- they seem fine.  The DeepSource analysis results are no longer available so we have no idea what the truncation issue may have been.  Closing.

closed 2023-01-31 by dgryski