tinygo-org/tinygo #1700: Full support for Digispark (ATTiny85)

erazemk, 2021-03-09:
I'm opening this issue because I want to work on adding full support for the Digispark board. From [the website](https://tinygo.org/microcontrollers/digispark/) I can see that only the GPIO pins are currently supported, so I want to help add support for everything else.

Since I have relatively little experience with programming microcontrollers I'd appreciate any help with where to start, but I was thinking of checking out the code that [Digispark have released for the Arduino IDE](https://github.com/digistump/DigistumpArduino/tree/master/digistump-avr) and try to port that to TinyGo.

aykevl, 2021-03-11:
That would be great!
Is there something you would like to work on? (Asking because some APIs might soon change, such as PWM).

erazemk, 2021-03-11:
> Is there something you would like to work on?

Nothing specific, but I can start with UART.

erazemk, 2021-03-11:
The Digispark [boards.txt](https://github.com/digistump/DigistumpArduino/blob/111396ffe5163c3613b9315cbea1a3eabc9ba6dc/digistump-avr/boards.txt#L16) file says the default frequency for the Digispark board is 16.5mhz (which is also confirmed if checked in the Arduino IDE), but TinyGo's [board_digispark.go](https://github.com/tinygo-org/tinygo/blob/138befd8a25d2a5c65e2e740038a8c3a44e95222/src/machine/board_digispark.go#L7) says it's 16mhz. Is there a specific reason for this?