tinygo-org/tinygo #1703: Add arm64 release for Linux/.deb

LaPingvino, 2021-03-10:
I use a Chromebook with Linux in its VM and it uses the ARM64 architecture. It would be very nice to be able to download a precompiled tinygo release for this.

aykevl, 2021-03-10:
Yes, that would be great but unfortunately CircleCI (where TinyGo is tested and release builds are built) does not have the necessary support. For more information: https://ideas.circleci.com/images/p/support-for-arm-based-docker-images

Hopefully we can support this in the near future, for now it is unfortunately not easily possible.

LaPingvino, 2021-03-10:
A recent update to that thread about 5 days ago suggests a usable preview should be available, and less than 24 hours ago seemingly that post has been further picked up.

LaPingvino, 2021-03-10:
Probably good then to keep this up as a tracking issue and to show the need.

briancampo, 2021-11-02:
I am also very interested in this for the same reasons.  
Any updates or progress?

loeffel-io, 2022-03-22:
Oh no, want to use tinygo on my arm so bad

niaow, 2022-11-04:
Closing as arm64 packages are included in releases v0.24.0 and up.

closed 2022-11-04 by niaow