tinygo-org/tinygo #1706: TinyGo does not work with Go installed from Snap

aykevl, 2021-03-11:
Some people are reporting TinyGo does not work with Go installed from Snap.

See: https://github.com/tinygo-org/tinygo/issues/433#issuecomment-671953170

aykevl, 2021-03-11:
Can someone (who runs Ubuntu and has Go installed from Snap) please share the output of the following commands?

```
which go
go env
```

Nerzal, 2021-03-13:
Just installed snap version :) 

To everyone reading this: i do not recommend using the Go snap installation, even for normal use. I had lots of problems with this and recommend to just install Go using download and unzipping method, like it is described on the Go website.

```bash
$ which go
/snap/bin/go
```

```bash
$ /snap/bin/go env
GO111MODULE="auto"
GOARCH="amd64"
GOBIN=""
GOCACHE="/home/tobias/.cache/go-build"
GOENV="/home/tobias/.config/go/env"
GOEXE=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GOINSECURE=""
GOMODCACHE="/home/tobias/go/pkg/mod"
GONOPROXY="dev.azure.com"
GONOSUMDB="dev.azure.com"
GOOS="linux"
GOPATH="/home/tobias/go:/home/tobias/go/src/github.com/tinygo-org/tinygo"
GOPRIVATE="dev.azure.com"
GOPROXY="https://proxy.golang.org,direct"
GOROOT="/snap/go/7013"
GOSUMDB="sum.golang.org"
GOTMPDIR=""
GOTOOLDIR="/snap/go/7013/pkg/tool/linux_amd64"
GCCGO="gccgo"
AR="ar"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD="/home/tobias/go/src/github.com/PacktPublishing/Programming-Microcontrollers-and-WebAssembly-with-TinyGo/go.mod"
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-g -O2"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build869670428=/tmp/go-build -gno-record-gcc-switches"
```

aykevl, 2021-03-13:
Thank you!
Is `/snap/bin/go` maybe a symlink to `/snap/go/7013/bin/go`? Is `/snap/go/7013` accessible from outside the snap package? TinyGo needs to find the actual GOROOT somehow, if you know how to do that you could see whether it is fixable here:

https://github.com/tinygo-org/tinygo/blob/138befd8a25d2a5c65e2e740038a8c3a44e95222/goenv/goenv.go#L151-L161

I'd have to install a VM to test this, which I'd rather avoid if that's not needed (I already tried LXC and WSL2, both don't work due to the special requirements of snapd).

Nerzal, 2021-03-14:
I'll need some days, as I'm a bit busy, with another project, but then i could check that out, yes. 
It does not seem to be to complicated, so i might just come up with a PR :)

aykevl, 2021-03-14:
That would be great! But take your time.

dmlambea, 2021-04-02:
I've been successfully using TinyGo with Go installed from Snap for a while. Since it might be my setup and I currently have time to send you the information you requested for helping the others, here it is:

```
$ go version
go version go1.15.10 linux/amd64
```

```
$ which go
/snap/bin/go
```

```
$ /snap/bin/go env
GO111MODULE=""
GOARCH="amd64"
GOBIN=""
GOCACHE="/home/dmlambea/.cache/go-build"
GOENV="/home/dmlambea/.config/go/env"
GOEXE=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GOINSECURE=""
GOMODCACHE="/home/dmlambea/go/pkg/mod"
GONOPROXY=""
GONOSUMDB=""
GOOS="linux"
GOPATH="/home/dmlambea/go"
GOPRIVATE=""
GOPROXY="https://proxy.golang.org,direct"
GOROOT="/snap/go/current"
GOSUMDB="sum.golang.org"
GOTMPDIR=""
GOTOOLDIR="/snap/go/current/pkg/tool/linux_amd64"
GCCGO="gccgo"
AR="ar"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD=""
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-g -O2"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build609128721=/tmp/go-build -gno-record-gcc-switches"
```

```
$ ls -la /snap/bin/go
lrwxrwxrwx 1 root root 13 mar 28 11:09 /snap/bin/go -> /usr/bin/snap
```

```
$ ls -la /snap/go/current
lrwxrwxrwx 1 root root 4 mar 28 11:09 /snap/go/current -> 7215
```

```
$ ls -la /snap/go/7215
total 168
drwxr-xr-x 12 root root   445 mar 16 04:07 .
drwxr-xr-x  4 root root  4096 mar 28 11:09 ..
drwxr-xr-x  2 root root   339 mar 16 04:07 api
-rw-r--r--  1 root root 55669 mar 11 17:14 AUTHORS
drwxr-xr-x  2 root root    38 mar 16 04:07 bin
-rwxr-xr-x  1 root root    41 mar 16 04:07 command-gofmt.wrapper
-rwxr-xr-x  1 root root    38 mar 16 04:07 command-go.wrapper
-rw-r--r--  1 root root  1339 mar 11 17:14 CONTRIBUTING.md
-rw-r--r--  1 root root 95475 mar 11 17:14 CONTRIBUTORS
drwxr-xr-x  7 root root   853 mar 16 04:07 doc
-rw-r--r--  1 root root  5686 mar 11 17:14 favicon.ico
-rwxr-xr-x  1 root root    71 mar 16 04:07 gofmtwrapper
-rwxr-xr-x  1 root root    68 mar 16 04:07 gowrapper
drwxr-xr-x  3 root root    27 mar 16 04:07 lib
-rw-r--r--  1 root root  1479 mar 11 17:14 LICENSE
drwxr-xr-x  3 root root    43 mar 16 04:07 meta
drwxr-xr-x 12 root root   198 mar 16 04:07 misc
-rw-r--r--  1 root root  1303 mar 11 17:14 PATENTS
drwxr-xr-x  5 root root    73 mar 16 04:07 pkg
-rw-r--r--  1 root root  1607 mar 11 17:14 README.md
-rw-r--r--  1 root root    26 mar 11 17:14 robots.txt
-rw-r--r--  1 root root   397 mar 11 17:14 SECURITY.md
drwxr-xr-x  3 root root    73 mar 16 04:07 snap
drwxr-xr-x 47 root root  1154 mar 16 04:07 src
drwxr-xr-x 23 root root  6198 mar 16 04:07 test
-rw-r--r--  1 root root     9 mar 11 17:14 VERSION
```

Hope it helps. If you need anything else I can provide, just ask.

oflebbe, 2021-04-02:
Is tinygo supported to be used with gccgo?

Nerzal, 2021-04-02:
@dmlambea yes, it helps! It is another indicator, that the PR i opened should fix the issue :) 
As the go root points to /snap/go/current, which simply is a link to the current install of go

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

PR #1729 (2021-04-15): goenv: Support Go snap installations by finding the goroot
I guess this does not need many words :) 

Fixes #1706 

Tested it and i can now build & flash code using TinyGo with a snap installation. 
Also tinygo version output looks good now :) 

```bash
$ which go
/snap/bin/go
```

```bash
./my_tinygo_18 version
tinygo version 0.18.0-dev linux/amd64 (using go version go1.16.2 and LLVM version 11.1.0)
```

current seems to always link on the currenty installed version
```bash
$ ls -lra /snap/go/
insgesamt 8
lrwxrwxrwx  1 root root    4 Mär 19 16:17 current -> 7221
drwxr-xr-x 12 root root  390 M [... cut]
closed 2021-05-12 by deadprogram