tinygo-org/tinygo #1713: targets/qtpy: incorrect volume name for UF2 mass storage device (MSD)

sago35, 2021-03-14:
I got the QTPy, but the volume name is different

I tried the following two bootloaders, but both had `QTPY_BOOT` instead of `QTPYBOOT`.

Is there a version where the Volume Name is QTPYBOOT?
In that case, we might want to extend the JSON to make multiple choices.

```
UF2 Bootloader v1.23.1-adafruit.1-328-gf06693a-dirty SFHWRO
Model: QT Py M0
Board-ID: SAMD21E18A-QTPy-v0
```

```
UF2 Bootloader v3.13.0 SFHWRO
Model: QT Py M0
Board-ID: SAMD21E18A-QTPy-v0
```

It seems to be QTPY_BOOT, at least in the latest bootloader.
https://github.com/adafruit/uf2-samdx1/blob/master/boards/QTPy_m0/board_config.h#L8

deadprogram, 2021-03-14:
We should change to `QTPY_BOOT` I think that was error on my part.

sago35, 2021-03-14:
OK.
I'll create a PR later.

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

closed 2021-05-12 by deadprogram