tinygo-org/tinygo #1722: Docker install drivers neeed custom commands

ricardojlrufino, 2021-03-15:
Using docker i need change some things to make work
Install drivers using:
> docker run --rm -v "/go:/gohost" -e "GOPATH=/gohost" golang:alpine go get tinygo.org/x/drivers

Next, move directories:
> mv /go/pkg/mod/tinygo.org/ /go/src
> mv /go/src/tinygo.org/x/drivers@v0.15.1/ /go/src/tinygo.org/x/drivers/

Then a use example: https://github.com/tinygo-org/tinygo-zoo
make build-arduino-colorlamp

Did I do something wrong? why doesn't he install the drivers in the correct location?

aykevl, 2021-03-15:
I have little experience with Docker (maybe someone else can help?), but you're passing the `--rm` flag. I think that will remove the container after you've used it, so maybe you can remove it?

ricardojlrufino, 2021-03-15:
yes, i need only mounted path: "-v "/go:/gohost"
the libraries are saved but it looks like in a different folder structure
I know little about GO, but maybe it's related to the package model? GO111MODULE = on

deadprogram, 2024-12-02:
Closing due to age and since TinyGo use with Docker is a lot more mature now. Thanks!

closed 2024-12-02 by deadprogram