tinygo-org/tinygo #1725: How do I place a variable at a specified address?

sago35, 2021-03-17:
I am using atsame54.
I need to put the Message RAM for CAN at 0x20000000 to 0x2000FFFF, but I don't know how.
Fortunately, I don't use much RAM at the moment, so I happen to have it in the right place.

I would like to know if there is a way to do this.

P.1176
https://cdn-learn.adafruit.com/assets/assets/000/097/530/original/SAM_D5xE5x_Family_Data_Sheet_DS60001507F.pdf

> Note:  Message RAM can only be located in the first 64 KB area of the system RAM.

deadprogram, 2021-03-17:
I think something like this is what you are looking for?

https://github.com/tinygo-org/tinygo/blob/release/src/runtime/runtime_atsamd21.go#L49-L62

sago35, 2021-03-30:
> would it be good enough if the memory was statically allocated (as a global)?

yes.
In this case, it is enough to place only global variables.

> I think something like this is what you are looking for?
> https://github.com/tinygo-org/tinygo/blob/release/src/runtime/runtime_atsamd21.go#L49-L62

It might also be possible to do it this way.
Either way, you need to change the linker script so that no other RAM is placed between 0x20000000 and 0x2000FFFF.

aykevl, 2021-03-30:
Please take a look here: https://github.com/tinygo-org/tinygo/pull/1759

It allows setting the section name of a global. With this and a small change to the linker script, you can place these special globals in the first 64kB of RAM.

deadprogram, 2024-11-27:
Closing since this question appears to have been answered. Thanks!

closed 2024-11-27 by deadprogram