tinygo-org/tinygo #1737: Arduino Mega 2560's SPI MISO/MOSI Flipped

soypat, 2021-03-21:
[https://github.com/tinygo-org/tinygo/blob/release/src/machine/machine_atmega2560.go#L136](The lines in question).
`sdi` and `sdo` seem to be flipped.

It would seem that the correct way of arranging `SPI0` according to [arduino's documentation](https://www.arduino.cc/en/reference/SPI) would be the following

```go
// SPI configuration
var SPI0 = SPI{
	spcr: avr.SPCR,
	spdr: avr.SPDR,
	spsr: avr.SPSR,
	sck:  PB1,
	sdi:  PB2,
	sdo:  PB3,
	cs:   PB0}
```

I've tested both ways, the second one seems to actually work.

soypat, 2021-03-23:
If someone can get a test jig on the Arduino Mega that'd be great. I couldn't get the SPI to work to save my life.

soypat, 2021-03-27:
I have gotten around to testing the Arduino Mega's SPI with the MCP3008 library. 

The following setup worked for me:
* Use TinyGo library without modification (works fine right now)
* Connecting Pin 51 to MCP3008 `Din` (MOSI)
* Connecting Pin 52 to MCP3008 `Dout` (MISO)
`machine.D51` is equal to `machine.PB2` (equal to `10`), which appears as `sdo` in `SPI0` configuration, which confused me (I thought SDO was Slave Device Output). I'd like to point out these names are easy to mistake, as manufacturers commonly use SDO/SDI for their pins.

deadprogram, 2021-03-28:
We try to follow the guidelines of the Open Source Hardware Association regarding naming of pins in the SPI interface. 

Please see:
https://www.oshwa.org/a-resolution-to-redefine-spi-signal-names/

Thank you!

closed 2021-04-10 by soypat