tinygo-org/tinygo #1747: Error mapping of registers for in-line asm

ewwaller, 2021-03-26:
I think this is a bug in the way inline assembly is processed.  I have some code that works properly with the _-opt 1_ flag, but does not work with the default optimization.  The following minimized example includes code from the ws2812 driver.  Running on a Adafruit Feather M4, this code displays increasing brightness of red, then green, then blue on the smart LED on that board (AKA Neopixel)  

This is done by transmitting three bytes, each encoded as 8 pulses (for a total of 24 pulses) to the device.  A narrow pulse codes a 0, and a wide pulse codes a 1

When it works (compiled with -opt 1) , the (correct) output waveform looks like this:  

[image]

But, with no optimization directive, it looks like this:
[image]

In this case, there are initial pulses for each of the three bytes, but the output never goes high for subsequent bits.  The time between bytes is the same as had been for the working example.

The tinygo go is :

```go
package` main

import (
	"device/arm"
	"image/color"
	"machine"
	"time"
)

func main() {
	const Period uint8 = 10
	var c int
	var r, g, b uint8
	var theColors [1]color.RGBA
	thePixel := Device{machine.D8}

	for {
		theColors[0] = color.RGBA{
			R: r,
			G: g,
			B: b,
		}
		thePixel.WriteColors(theColors[:])
		time.Sleep(time.Duration(Period) * time.Millisecond)
		switch c {
		case 0:
			r++
			if r == 0 {
				c++
			}
		case 1:
			g++
			if g == 0 {
				c++
			}
		case 2:
			b++
			if b == 0 {
				c++
			}
		default:
		}
		c = c % 3
	}
}

// Device wraps a pin object for an easy driver interface.
type Device struct {
	Pin machine.Pin
}

// Write the raw bitstring out using the WS2812 protocol.
func (d Device) Write(buf []byte) (n int, err error) {
	for _, c := range buf {
[... 388 lines cut ...]
    5d86:       bf00            nop
    5d88:       bf00            nop
    5d8a:       bf00            nop
    5d8c:       bf00            nop
    5d8e:       bf00            nop
    5d90:       bf00            nop
    5d92:       bf00            nop
    5d94:       bf00            nop
    5d96:       bf00            nop
    5d98:       bf00            nop
    5d9a:       bf00            nop
    5d9c:       bf00            nop
    5d9e:       bf00            nop
    5da0:       bf00            nop
    5da2:       bf00            nop
    5da4:       bf00            nop
    5da6:       bf00            nop
    5da8:       bf00            nop
    5daa:       f1bc 0c01       subs.w  ip, ip, #1
    5dae:       d18a            bne.n   5cc6 <(main.Device).WriteByte+0x1c>
    5db0:       f381 8810       msr     PRIMASK, r1
    5db4:       4770            bx      lr

```
Note that in this case, the optimizer figured out it did not need to call machine as the information was available at compile time, so it used constants.  The problem appears to be, in this second case, it is attempting to use the ip register for both the count down from 8 to 0 [ see 0x5daa - 0x5dae], **AND** as the pointer to the register that contain the I/O register [see 0x5cf8 and 0x5d4a]

aykevl, 2021-03-26:
I can confirm the issue. I think the issue is that `maskSet`, `maskClear` and `i` are all the constant 8 and the optimizer recognizes this and puts them all in the same register. That would be fine, if the values were indeed treated as constants. But while `maskSet` and `maskClear` are constant (and merging them is a good idea), `i` is not. This leads to the miscompilation.

Right now the inline assembly format does not support marking certain input values as clobbered. I'm not sure whether I really want to add such support, `AsmFull` is already pretty complicated. Instead, it might be better to move these snippets of inline assembly to C (usable through CGo) so that all the features of GCC style inline assembly can be used (including marking some inputs as clobbered).
CGo might sound expensive, but in TinyGo it's just a regular function call and I have plans of optimizing it further so that it is optimized together with Go code (and can be inlined, etc, see #1707 for some related work).

aykevl, 2021-03-26:
Also, I just want to say thank you for this super helpful bug report. It's a very clear description of the problem, with a reduced code sample even.

ewwaller, 2021-03-26:
Thanks, and my pleasure.   I am new to go, but have spent my professional career with microcontrollers and embedded systems.  I am new to go, but am increasingly excited about tinygo.

ewwaller, 2021-04-02:
I have been playing with this, and have hit a roadblock with CGo and could use some help getting back on the tracks.

Here is my code for the ws2812 driver:

"""// +build atsamd51

package ws2812

// This file implements the WS2812 protocol for 120MHz Cortex-M4
// microcontrollers.
// Note: This implementation does not work with tinygo 0.9.0 or older.

import (
	"device/arm"
)

/*
#include <stdint.h>
void writebyte( unsigned char c, \
                uint32_t maskSet, \
                uint32_t maskClear, \
                uint32_t* portSet, \
                uint32_t* portClear){
   int cnt;
    int i;
    for (i = 7 ; i != 0xff ; i--) {
        if ((1<<i) & c) {
            cnt=4;
            while (cnt--);
            *portClear = maskClear;
            cnt=8;
            while (cnt--);
        } else {
            cnt=8;
            while (cnt--);
            *portClear = maskClear;
            cnt=4;
            while (cnt--);
        }
        *portSet = maskSet;
        *portClear = maskClear;
    }
 }
*/
import "C"

// Send a single byte using the WS2812 protocol.
func (d Device) WriteByte(c byte) error {
	// For the Cortex-M4 at 120MHz
	portSet, maskSet := d.Pin.PortMaskSet()
	portClear, maskClear := d.Pin.PortMaskClear()
	// See:
	// https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/
	// T0H: 32-34   cycles or  266.67ns -  283.33ns
	// T0L: 101-103 cycles or  841.67ns -  858.33ns
	//   +: 133-137 cycles or 1108.33ns - 1141.67ns
	// T1H: 73-75   cycles or  608.33ns -  625.00ns
	// T1L: 58-60   cycles or  483.33ns -  500.00ns
	//   +: 131-135 cycles or 1091.67ns - 1125.00ns

	mask := arm.DisableInterrupts()
	C.writebyte(
		C.uchar(c),
		C.uint32_t(maskSet),
		C.uint32_t(maskClear),
		portSet,
		portClear)
	arm.EnableInterrupts(mask)
	return nil
}
"""

but, I get this when I build it:

"""ewaller@odin/home/ewaller/devel/go/src/ewaller.local/featherBlink[1] % tinygo build -target=feather-m4 -size full  -o test.uf2 
ld.lld: error: undefined symbol: writebyte
>>> referenced by ws2812_m4_120m.go:58 (/home/ewaller/devel/go/src/ewaller.local/featherBlink/drivers/ws2812/ws2812_m4_120m.go:58)
>>>               /tmp/tinygo887064602/main.o:((featherBlink/drivers/ws2812.Device).WriteByte)
error: failed to link /tmp/tinygo887064602/main: exit status 1
ewaller@odin/home/ewaller/devel/go/src/ewaller.local/featherBlink[1] % 
"""
It compiles, but dies at the linker.  Not sure where to go from here.

deadprogram, 2024-12-14:
Hello @ewwaller @aykevl just wondering if this issue is still valid or it can be closed? 

https://github.com/tinygo-org/drivers/blob/release/ws2812/ws2812-asm_cortexm.go would seem to indicate that this is no longer a problem.

Thanks!

aykevl, 2024-12-14:
Yeah this code was essentially rewritten and the compiler now knows the `i` parameter will be modified and therefore needs a dedicated register (using `[i]"+r"(i)`). So it should be fixed.

closed 2024-12-14 by aykevl

ewwaller, 2024-12-14:
I've not played with it for quite some time.   Let me check into it.